raspberrypi/rpi-imager #480: writing stuck at 99%

CarloCattano, 2022-09-10:
raspberry pi imager 1.7.2 is writing to the sd card and getting stuck at 99%

OS: Manjaro Linux x86_64 
Kernel: 5.19.1-3-MANJARO 

If selecting my own image downloaded from the page It freezes at 99%
However when choosing the OS from the imager interface it works

maxnet, 2022-09-10:
If you grab a coffee, and wait a while, it will not complete eventually?

Progress indication can indeed be inaccurate if it's a compressed local image that is not a .zip.
We need to know the uncompressed image size for that and that is not available in all compression formats.

CarloCattano, 2022-09-10:
Hm thanks. No, not really. It just hangs forever and its weirdly fast in writing. When selecting the Os from the interface is much slower but it finishes and verifies.
Just wanted to report this but unfortunately I didn't see any useful info on my logs

maxnet, 2022-09-10:
>No, not really. It just hangs forever and its weirdly fast in writing.

Weirdly fast in progress indication is normal, if we do not know the uncompressed file size.
It will then show the progress indication READING the image file instead.
Once everything is read/uncompressed, and the data to be written is in the kernel's write buffer it will show 99%, 

That it does not complete eventually do is weird though.

lurch, 2022-09-10:
If you manually uncompress your locally-downloaded image before writing it, does that work any better?

CarloCattano, 2022-10-18:
Didn't have time to try this solution but i will. Seems that we know where to look now. Plenty of ram on my end, as on an M1 too, so that's discarded

ejgardner413, 2023-01-22:
I was extremely patient. It was stuck at 99% using the compressed Ubuntu server image for over 20 minutes. It eventually completed.

maxnet, 2023-01-25:
>Same happening to me on OSX Mac M1.
--
>I was extremely patient. It was stuck at 99% using the compressed Ubuntu server image for over 20 minutes. It eventually completed.
--

[image]

Ubuntu's image does have more than 1 GB of excess space at the end.
If using a downloaded .xz file, it may be at 99% for some time, while it is writing out that, because it uses how far it is READing the compressed input file for progress percentage in that case, and zeroes ("empty" space) compress very well...

If using an image from the repository this issue does not occur, as it is able to get metadata about the uncompressed image size from our server, and can use that in combination with the amount WRITTEN for better progress reports.

For reference:

This is how long Ubuntu Server should take with a Samsung SD card and USB 3 card reader, on a M2 Mac Mini.

From Internet: https://www.youtube.com/watch?v=65K_rkmpr8M
From local file: https://www.youtube.com/watch?v=Avjo30fJHHc

Yes, of the 3 minutes it takes to write the file, it is stuck for 30 seconds at 99% in the last case. While it does not occur when using the repository.
If it really takes 20 minutes in your case, there may be something else wrong with your hardware.

That do is with the stock Ubuntu Server file. 
If you took your own image of say a 64 GB SD card, and there is a lot of highly compressable emptyness at the end, it naturally do may take significantly longer.

lurch, 2023-01-25:
> If using an image from the repository this issue does not occur, as it is able to get metadata about the uncompressed image size from our server, and can use that in combination with the amount WRITTEN for better progress reports.

For the benefit of people reading this issue and wondering what the difference is... the "problem" with the `.xz` file format is that it doesn't store the size of the uncompressed data. So if you have a 500MB `.xz` file, you have no way of knowing whether that'll uncompress to 600MB of (not-very-compressible) data, or 8GB of (highly-compressible) data... which is why it's impossible to give a reliable progress-indicator without external metadata.

maxnet, 2023-01-25:
>the "problem" with the .xz file format is that it doesn't store the size of the uncompressed data.

Well, it may have, but doing anything with that currently lacks implementation...

(there is a "backwards size" stored in the footer at the end of the stream.
but there is the corner case that there can be more than one stream, in which case it is less trivial to extract without parsing blocks... Lot of work for something the average user that is just using our repository will not benefit from.
https://tukaani.org/xz/xz-file-format-1.1.0.txt )

lurch, 2023-01-26:
Ah! Apologies for the misinformation - perhaps I was getting confused with some other archive / compression format then :shrug:

colinwang1703, 2023-06-27:
Just wait.

closed 2023-06-27 by CarloCattano

lurch, 2025-11-26:
The recently-released Imager 2.0 changes this behaviour to periodically sync during writing, rather than saving it all up till the end. See https://github.com/raspberrypi/rpi-imager/commit/8cdba8759eac2ab6e859d2444f2b1a3ba6c06ca4

P-Ramos16, 2025-11-26:
Nice, I have to write two RPi 5s today so I will try one before updating rpi-imager and one after. Should take almost the same mmount of time tho, writting to disk is the bottleneck