raspberrypi/rpi-imager #481: Reset advanced options

golemelog, 2022-09-12:
Selecting "use always" there's no way to reset, nor automatically fill "wifi in use" again.
I'm wondering how can I reset the settings (on Mac) so I can start from scratch. 
Remove the app with uninstaller and reinstalling did not work.  Is there a settings file I can delete?

maxnet, 2022-09-12:
>I'm wondering how can I reset the settings (on Mac) so I can start from scratch.

Changing it from "use always" to "for this session only", pressing "save", and restarting Imager afterwards should work.

DevonTomatoSandwich, 2022-10-23:
I had a similar problem, I did that ^, all my settings got wiped. I'm on MacOS also so it's partly my fault lol

DevonTomatoSandwich, 2022-10-23:
[image]
After i close and open the app, the wifi password has some weird hash at the start

maxnet, 2022-10-23:
>After i close and open the app, the wifi password has some weird hash at the start

That is as intended.
Wifi passwords are stored as PSK2 hash (taken over SSID and password. So if your SSID changes, you need to re-enter password as well).

AlAmantea, 2022-12-09:
I have an issue with this as well.
Imager tries to write the password incorrectly on subsequent writes and does not remove the hash from the password before writing.
I have to manually reenter the wifi password on every image. This feature is definitely not working as intended.

maxnet, 2022-12-09:
>Imager tries to write the password incorrectly on subsequent writes
>and does not remove the hash from the password before writing.

Do not understand what you mean.

- Hashing is an irreversible one-way process. There is no way to "remove the hash", once applied.
- wpa_supplicant accepts PSK2 hashed passwords fine. So there is no need for a password without hash.

>If I set the feature "This session only" the hash does not appear in the password and works correctly.

You will not see it in the user-interface in that case, but it is certainly applied.
Open the file "firstrun.sh" on the FAT partition of the SD card in a text editor to see.

Does the "firstrun.sh" look different on first and subsequent runs?

tdewey-rpi, 2023-10-19:
Closing due to inactivity