raspberrypi/rpi-imager #483: Imager does not find SD Card exposed to Virtual Box VM

serkonda7, 2022-09-15:
I followed this tutorial to expose a SD card to a VM: https://scribles.net/accessing-sd-card-from-linux-virtualbox-guest-on-windows-host/
However the app does not show the card as storage device, so how do I flash it now?

Output of `fdisk -l`:
```
Disk /dev/sdb: 7,4 GiB, 7948206080 bytes, 15523840 sectors
Disk model: VBOX HARDDISK   
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0xf4673170

Device     Boot  Start      End  Sectors  Size Id Type
/dev/sdb1         8192   532479   524288  256M  c W95 FAT32 (LBA)
/dev/sdb2       532480 15523839 14991360  7,1G 83 Linux
```

maxnet, 2022-09-15:
>I followed this tutorial to expose a SD card to a VM: https://scribles.net/accessing-sd-card-from-linux-virtualbox-guest-on-windows-host/

Your tutorial shows how to make the device appear as SATA drive to the VM.
We do not show internal SATA drives in Imager, as people generally do not want to use Imager to overwrite the files on their hard drive...

Try making your SATA drive "hot pluggable", so it is no longer treated as an internal drive, or attach it to a different virtual controller type such as USB:

[image]

lurch, 2022-09-15:
...or you could just ignore the whole VirtualBox part, and just run RPi Imager natively on Windows?? :confused:

serkonda7, 2022-09-15:
@maxnet thanks for the explanation.
Marking it as hot-pluggable had no effect but changing the controller to USB seems to be fine :)

> ...or you could just ignore the whole VirtualBox part, and just run RPi Imager natively on Windows??

@lurch (sadly) not possible due to corporate restrictions.

closed 2022-09-15 by serkonda7

lurch, 2022-09-15:
Seems kinda bizarre that you can't run RPi Imager natively, but you can run (whatever you like?) inside VirtualBox? Meh :shrug: :wink: