raspberrypi/rpi-imager #484: Windows Build Missing dlls

jay-m-dev, 2022-09-17:
**Issue Description**
Using Qt 5.15.2 MinGW 32 bit on Windows 11, Qt creator finishes building successfully, but compiling the NSIS script (rpi-imager.nsi) displays errors regarding the dll files in the **deploy** directory.
The error is on line 222 of the rpi-imager.nsi. (I'm attaching my build code and the output.log from MakeNSISW)
What I've been able to gather is that the **deploy** directory should have the dll files after building in Qt Creator, but as you can see from the attached build file, the **deploy** directory is empty.

Is there something I'm missing on the build step?

**Expected Behavior**
Generate rpi-imager.exe after compiling the rpi-imager.nsi script.

[MakeNSISW_output.log](https://github.com/raspberrypi/rpi-imager/files/9590725/MakeNSISW_output.lo
[build-src-Desktop_Qt_5_15_2_MinGW_32_bit-Debug.zip](https://github.com/raspberrypi/rpi-imager/files/9590726/build-src-Desktop_Qt_5_15_2_MinGW_32_bit-Debug.zip)
g)

maxnet, 2022-09-17:
>Qt creator finishes building successfully

Can you attach the build log (of cmake/qt creator, not nsis)?

Copying stuff to "deploy" is supposed to be part of the build process. ( https://github.com/raspberrypi/rpi-imager/blob/qml/src/CMakeLists.txt#L208 )

jay-m-dev, 2022-09-17:
Here are my logs. It looks like I'm having issues with dependencies. 
[cmake.txt](https://github.com/raspberrypi/rpi-imager/files/9592111/cmake.txt)
[compile_output.txt](https://github.com/raspberrypi/rpi-imager/files/9592110/compile_output.txt)

maxnet, 2022-09-18:
Seems to be mixing include files from two instances of Mingw on the system? 

C:/Qt/Tools/mingw810_32/i686-w64-mingw32/include and C:/MinGW/include

Did you perhaps put c:/mingw in PATH or other global environment variables at some point?
If so get rid of that...

jay-m-dev, 2022-09-19:
Thank you so much for your help!
I tried with a fresh install, using only the MinGW81_32 version that comes with Qt and it works now!
I'd also like to mention that I used the OpenSSL version that installs with Qt (C:\Qt\Tools\OpenSSL) and this worked for me.
Thank you!

closed 2022-09-19 by jay-m-dev

maxnet, 2022-09-19:
>I'd also like to mention that I used the OpenSSL version that installs with Qt (C:\Qt\Tools\OpenSSL) and this worked for me.

Recall that that OpenSSL version was compiled with Visual C++ instead of mingw.

If you are compiling for your own use, that should not be a problem.
But if you intend to distribute to others, be aware that it may only work on systems that have VC++ runtime .dll's installed.
(Could be that those are installed by default on modern Windows versions nowadays, but would not rely on that either).