raspberrypi/rpi-imager #486: Apostrophe in SSID generates an invalid firstrun.sh

mjc, 2022-09-22:
`firstrun.sh` does not correctly escape single quote in WiFi SSIDs, leading to a pi that simply powers off on first boot.

There's a few other similar things that shellcheck catches in there, also.

rbsamoht, 2022-10-12:
Same issue here, using a password in the SSID will generate an invalid bash output in the `firstrun.sh` file. The apostrophe is not properly escaped:
```bash
'mySSIDWith\'Apostrophe'
```
Something like `'\''` should probably be used instead:
```bash
'mySSIDWith'\''Apostrophe'
```
## Environment
- Windows 11
- Raspberry Pi Imager v1.7.3

## Current workaround
Downgrade to Raspberry Pi Imager v1.7.2, that doesn't seem to be affected by this issue

tomstuart, 2022-10-15:
This bit me too. My workaround was to manually edit line 42 in the generated `firstrun.sh` to use double (vs single) quotes around the SSID.

tomstuart, 2022-10-15:
The problem was introduced in 4961990ce00892325b08a10d91f5eba9b3711e7c. It [tries to use](https://github.com/raspberrypi/rpi-imager/commit/4961990ce00892325b08a10d91f5eba9b3711e7c#diff-8625d58f57ae280809d073a633e766757ea0f518788ce2b7d7eceb854f3d7ee9R717-R719) the [`escapeshellarg` function](https://github.com/raspberrypi/rpi-imager/blob/v1.7.3/src/OptionsPopup.qml#L572-L574) to escape the SSID, but that function doesn’t escape single quotes correctly for Bash ([“A single quote may not occur between single quotes, even when preceded by a backslash.”](https://www.gnu.org/software/bash/manual/bash.html#Single-Quotes)). As @rbsamoht says [above](https://github.com/raspberrypi/rpi-imager/issues/486#issuecomment-1276511791), it’d be better to replace `'` with `'\''` so that the string is split into two parts which are joined with an unquoted apostrophe.

livvy94, 2022-10-28:
I just ran into this! For the googlers out there, it results in this error:
```
[   23.217672] firstrun.sh[309]: /boot/firstrun.sh: line 70: unexpected EOF while looking for matching '"'
[   23.219300 firstrun.sh[309]: /boot/firstrun.sh: line 70: syntax error: unexpected end of file
```
At this point, it powers down.
```
[  OK  ] Reached target Power-Off.
```

Longsight, 2022-11-12:
> The problem was introduced in [4961990](https://github.com/raspberrypi/rpi-imager/commit/4961990ce00892325b08a10d91f5eba9b3711e7c). It [tries to use](https://github.com/raspberrypi/rpi-imager/commit/4961990ce00892325b08a10d91f5eba9b3711e7c#diff-8625d58f57ae280809d073a633e766757ea0f518788ce2b7d7eceb854f3d7ee9R717-R719) the [`escapeshellarg` function](https://github.com/raspberrypi/rpi-imager/blob/v1.7.3/src/OptionsPopup.qml#L572-L574) to escape the SSID, but that function doesn’t escape single quotes correctly for Bash ([“A single quote may not occur between single quotes, even when preceded by a backslash.”](https://www.gnu.org/software/bash/manual/bash.html#Single-Quotes)). As @rbsamoht says [above](https://github.com/raspberrypi/rpi-imager/issues/486#issuecomment-1276511791), it’d be better to replace `'` with `'\''` so that the string is split into two parts which are joined with an unquoted apostrophe.

Probably an even cleaner solution would be to prefix the string with `$`, forcing [ANSI-C quoting](https://www.gnu.org/software/bash/manual/html_node/ANSI_002dC-Quoting.html) and allowing `\'` to work as expected, eg:

`   /usr/lib/raspberrypi-sys-mods/imager_custom set_wlan 'Don\'tgonearthe Castle'` - fails
`   /usr/lib/raspberrypi-sys-mods/imager_custom set_wlan $'Don\'tgonearthe Castle'` - works

Though having said that I notice `escapeshellargs` has already been fixed with the `'"'"'` approach, which does work.

nathan-hook, 2022-12-11:
Just to dog pile a bit.  I have run into this problem as well and it has caused several days of futzing to troubleshoot the problem and has been quite frustrating.

In my particular case, I was lucky enough to see the messages that [livvy94](https://github.com/livvy94) posted and wound up here.

Two (fairly obvious) workarounds:
* Don't setup wifi when running the imager and then run `sudo raspi-config` once the RPI has completed booting.  But this causes the pi to either be connected to a monitor and keyboard or hard wired into a router.
* Use an older raspios image.  I ended up using the [2022-04-04-RASPIOS-BULLSEYE-ARM64-LITE.IMG](https://downloads.raspberrypi.org/raspios_lite_arm64/images/raspios_lite_arm64-2022-04-07/)

maxnet, 2022-12-11:
>Two (fairly obvious) workarounds:

Another option is to try 1.7.4
Not officially released yet, but you can grab a build here: [Windows](https://www.maxnet.eu/tmp/174-fat/imager-1.7.4.exe) [Linux](https://www.maxnet.eu/tmp/174-fat/rpi-imager_1.7.4_amd64.deb) [Mac](https://www.maxnet.eu/tmp/174-fat/Raspberry%20Pi%20Imager%201.7.4.dmg)

nathan-hook, 2022-12-11:
> Another option is to try 1.7.4 Not officially released yet, but you can grab a build here: [Windows](https://www.maxnet.eu/tmp/174-fat/imager-1.7.4.exe) [...]

I ended up deleting my comment because the second workaround actually didn't work.

But am happy to report that the 1.7.4 version of the imager seems to have fixed the problem.

vollkorn1982, 2022-12-13:
I ran into this problem as well and my solution was to change my SSID to the hex representation, similarly as is already done with the passphrase. This seemed the safest option to me, because no quoting mistakes can be made and it definitely avoids shell injection attacks or similar problems.

https://web.mit.edu/freebsd/head/contrib/wpa/wpa_supplicant/wpa_supplicant.conf says:
```
# ssid: SSID (mandatory); network name in one of the optional formats:
#	- an ASCII string with double quotation
#	- a hex string (two characters per octet of SSID)
#	- a printf-escaped ASCII string P"<escaped string>"
```

maxnet, 2022-12-13:
>and it definitely avoids shell injection attacks or similar problems.

Well, on Windows, OSX and Linux, any user (not just root/administrator) can access files on the FAT partition of removable drives.
So one does not need to (ab)use Imager to inject shell commands. Can simply open firstrun.sh in notepad and add them there. :-)

vollkorn1982, 2022-12-13:
Hi @maxnet, to be honest, your answer really worked me up. Let me explain why.

I gave a hint about how to build this feature more secure, but instead of approaching this possibility you brushed it away. Yes, your threat model may make this not a necessary security improvement, but other people's threat models may. I won't go into a discussion about under which conditions it could be a valid security issue. These discussions usually only waste time and energy magnitudes higher that it would've to simply implement the more secure way.

Instead, I'll let you know what your message communicated to me.

1. "Yeah, I know this is an insecure coding style, but I don't care." This is cultivating insecure coding habits. I now believe you're ok with building up technical debt.

2. I think this will tell other people working on this project, that it's fine to knowingly write insecure code as long as they find a plausible excuse not to do it proper. 

3. Brushing off a security advice like this may also signal to external people reading here, that it's probably pointless to tell you about their security concerns and findings, because they would have to convince you first, instead of you being thankful for helping to improve the project. With regards to security, this can be pretty bad.

Now, quite honestly, writing this post might already have taken longer than looking up how the hex encoding of the password is done and applying it to the SSID as well. So, maybe I am guilty of my own complaint, too, but maybe my message resonates with you and I made this world a little bit better.

I'll try to find the corresponding code and make a PR now.

maxnet, 2022-12-13:
>I'll try to find the corresponding code and make a PR now.

Your PR is unlikely to work without new RPI OS images...
Setting wifi password is outsourced to a /usr/lib/raspberrypi-sys-mods/imager_custom script inside RPI OS nowadays, which in turns call raspi-config and nmcli.
Wouldn't count on those accepting hexadecimal SSIDs right now.
(wpa_supplicant may accept them, but only if the value is without quotes in the configuration file, and don't think it ends up there that way).

And no, doing configuration by passing user input to scripts (that resulted in this issue in the first place) wasn't my personal preference either.
Wasn't my decision.

cillian64, 2023-11-07:
After https://github.com/RPi-Distro/raspberrypi-sys-mods/pull/83 `'` works in SSIDs and PSKs (as well as all the other interesting characters I could think of to try)