raspberrypi/rpi-imager #487: [Feature Request] macOS arm64 Support

robd003, 2022-09-23:
I noticed that rpi-imager doesn't support Apple SIlicon natively. Considering that we're almost 3 years since the launch of Apple Silicon could we get a build available for it?

My office has replaced all of the old Intel Macs with new Apple Silicon Macs this year.

Having both a x86_64 build (for legacy systems) and an arm64 build would be fantastic.

lurch, 2022-09-23:
See #353

ghollingworth, 2022-09-24:
Currently, the downloads from M1 Macs are only 0.1% of the total downloads, this is not enough to warrant including a new build.  You can still use the x86 version with Rosetta anyway

closed 2022-09-24 by ghollingworth

infinitepower18, 2022-10-03:
Are you still going to stick to your stance of not supporting Apple Silicon when Apple eventually removes Rosetta, just like they did with the PowerPC transition?

ghollingworth, 2022-10-03:
The original Rosetta sounds like it lasted for seven to eight years...  By then I'm sure the platform will have a lot more users, but at the moment 0.1% is not enough.

asyba, 2022-12-26:
> Currently, the downloads from M1 Macs are only 0.1% of the total downloads, this is not enough to warrant including a new build. You can still use the x86 version with Rosetta anyway

because ppl first check if Raspberry Pi Imager is compatible with M1 mac on google and when they found out that is not, they don't event ry to download  the app, so measuring 0.1% downloads is not a good tool ... lol