raspberrypi/rpi-imager #490: No wifi -  RPI 4  in RPI OS 32 BIT  when using a SSID and Password in Raspberry Pi Imager

goyetus, 2022-09-26:
Using the Raspberry Pi imager  and setting an SSID and Password:

RPI OS 32 bit  full  =  Wifi working.
RPI OS 32 bit lite  =  Wifi not working 

Error in RPI OS 32 lite = WIFI IS CURRENTLY BLOCKED BY RFKILL

[image]

XECDesign, 2022-09-27:
Looks okay on this end.

I see the kernel you have is older than what we ship with, so it doesn't seem to be the latest armhf lite image.

icemtel, 2022-11-08:
Only-slightly relevant question: Where does RP-imager writes the wi-fi configuration? I am debugging an issue on Ubuntu 22.04 where I am not able to change my wi-fi network.

icemtel, 2022-11-09:
@maxnet big thanks, I meant the ubuntu server. 

This is really off-topic now, but do you have any tips on switching the wi-fi network? `nmcli` gives `wlan0: unavailable`, `sudo wpa_cli` fails to save a config, and the advice from the internet did not resolve that.

maxnet, 2022-11-10:
>This is really off-topic now, but do you have any tips on switching the wi-fi network? 

Ubuntu Server has gone netplan.
It will consult the files in /etc/netplan on every boot and generate brand new configuration files from that. So if you modify configuration files in any other directory they will be overwritten.

For configuration examples see: https://netplan.io/examples
May need to toss a country setting like "cfg80211.ieee80211_regdom=GB" in /boot/firmware/cmdline.txt
A wifi country code setting did was added to upstream netplan recently, but I don't think it made it into the Ubuntu version you are using yet.

icemtel, 2022-11-11:
> For configuration examples see: https://netplan.io/examples

Great, this worked! Thank you for your help.

tdewey-rpi, 2025-12-16:
Closing this as long-stale, and it looks like all relevant questions were answered.

closed 2025-12-16 by tdewey-rpi