raspberrypi/rpi-imager #498: Raspberry pi  3b failed to boot with advanced settings.

flynnwang, 2022-10-30:
The new installation failed to boot when using the raspberry pi imager to install a new version of raspberry OS (bullseye) with advanced settings (hostname, username/password, ssh, language). After reinstalling the pi imager and installing the raspberry OS with no advanced settings, the issue was solved. 

* raspberry pi: 3b
* pi imager: v1.7.3
* pi imager is running on: macOS Monterey, Apple M1 Pro
* raspberry OS: bullseye (both lite and desktop)

[image]

maxnet, 2022-10-30:
Do you have a wifi SSID with an ' in it?

flynnwang, 2022-10-30:
> SSID

Yes, also set the Wifi SSID and its password.

maxnet, 2022-10-30:
>Yes, also set the Wifi SSID and its password.

What does your SSID look like?

Only normal characters (ABCD1234) or also special characters (like single quotes ' )?

flynnwang, 2022-10-31:
> > Yes, also set the Wifi SSID and its password.
> 
> What does your SSID look like?
> 
> Only normal characters (ABCD1234) or also special characters (like single quotes ' )?

Mine is like: "Wang's WiFi"

maxnet, 2022-10-31:
Duplicate of #499
(Fixed in git, but not yet in release)

flynnwang, 2022-10-31:
Good to know :)

closed 2022-10-31 by flynnwang