raspberrypi/rpi-imager #499: [OS Request] Could you please add MoodleBox?

martignoni, 2022-10-30:
[MoodleBox](https://moodlebox.net/) combines a wireless access point with a full featured Moodle server, on a Raspberry Pi. MoodleBox is used to provide learning and teaching with the [LMS Moodle](https:/moodle.org/) where infrastructure is non-existent or inadequate, for example in developing countries or where infrastructure has been destroyed, in the wake of war episodes or natural disasters.

Providing an official way to copy the MoodleBox image would enable people to get it easier.

See https://moodlebox.net/moodlebox-rpi-imager.json.

XECDesign, 2022-10-31:
It seems to ship with SSH pre-enabled with a known user:pass set and SSH host keys pre-installed. Is that intended?

martignoni, 2022-10-31:
Yes, it is. The public for this is not at all tech-savvy and installing Moodle needs an account anyway.

martignoni, 2022-11-01:
@XECDesign: BTW, is there any mean to return to a state without default user:pass after having configured a Pi?

XECDesign, 2022-11-01:
If I understood the question correctly:

Remove the password from the current account and lock it:
```
echo "yourusername:*" | sudo chpasswd -e
```
or:
```
sudo usermod --pass='*' yourusername
```
In `/etc/shadow` you should see that the "encrypted" password should now be `*`.

(Replace `yourusername` with whatever the preconfigured user you have is, probably `moodlebox` )

Then to re-enable the user rename mechanism.

If running as root:
```
SUDO_USER="yourusername" rename-user -f -s
```

If running as the user to be renamed:
```
sudo rename-user -f -s
```

The `-f` flag runs the full wizard with locale, wifi config and so on. I suspect in your case, you probably don't want this,

The `-s` flags tells the script not to restart the system after it runs. `-r` can be used to restart. If neither are specified, it will ask.

martignoni, 2022-11-01:
This would be a serious option when (if?) the image is included in RPi Imager default list.

XECDesign, 2022-11-01:
Yeah, there's no objection to including the image. MoodleBox seems very useful.

It's just the concern about having a wide open system by default. I think having an audience that's not tech-savvy, which might not understand the implications of having a box in their home or organisation that grants outside access to their network. Also, having the SSH host keys there opens it up to MITM attacks.

It would make sense it access was limited to the local subnet or wlan only, but AIUI, if the ethernet cable is plugged in, that's not the case.

I understand most people are behind a NAT and router firewall, but we've seen many people go out of their way to get themselves hacked.

martignoni, 2022-11-01:
> Also, having the SSH host keys there opens it up to MITM attacks.

Indeed, and this could also be suppressed thanks RPi Imager allowing to enable it when flashing the card.

Two more questions:
- Would this be a prerequisite for the MoodleBox image inclusion?
- Is there anything to do in the image to be able to use those mecanism (creating a user and SSH enabling)?

ghollingworth, 2022-11-01:
I'd like to say this is a prerequisite because we're trying to be better when it comes to security on Raspberry Pi.  But you are probably not alone in terms of current image suppliers who haven't also done this...  

But we do need to check with our other suppliers and ask them to also update their images to use the methods provided to handle this username/password/SSH type problem.

XECDesign, 2022-11-01:
> Is there anything to do in the image to be able to use those mecanism (creating a user and SSH enabling)?

Sorry, I'm not sure I'm parsing the question right. Could you please rephrase it?

martignoni, 2022-11-01:
Given I've a standard MoodleBox image (which is a standard RPi OS image with some installation and configuration) and I've removed the default user and disabled SSH, is there any other step to do so that RPi Imager mecanisms to create a user and enable SSH work (or will it work anyway)?

(Sorry, English isn't my first language.)

martignoni, 2022-11-01:
> I'd like to say this is a prerequisite because we're trying to be better when it comes to security on Raspberry Pi.

Ok, I'll implement this for the next version. Would it be OK if I release a RC of the new image so that you include it?

lurch, 2022-11-01:
IIRC the Raspberry Pi OS images come "partially unconfigured" and a lot of the initial configuration happens during first bootup. So rather than try to "restore" an already-booted OS image to this same initial state, you _might_ want to have a look at https://github.com/RPi-Distro/pi-gen (which is the tool that creates the Raspberry Pi OS images) and adapt it to create your MoodleBox images in this same initial state? :shrug:  (but I've not really played with pi-gen myself, so I've no idea how hard this would be.)

martignoni, 2022-11-01:
Thanks for your input. I've tried with pi-gen (see https://github.com/moodlebox/pi-gen-moodlebox), but unfortunately a few steps cannot be automated using it (eg. Moodle installation and configuration).

XECDesign, 2022-11-01:
> Given I've a standard MoodleBox image (which is a standard RPi OS image with some installation and configuration) and I've removed the default user and disabled SSH, is there any other step to do so that RPi Imager mecanisms to create a user and enable SSH work (or will that work anyway)?

Ah yeah, that should be possible. If you make a 'first attempt' at such an image and send me a link, I can take a look and tell you what extra steps would be needed.

martignoni, 2022-11-04:
@XECDesign: here's a first attempt: https://moodlebox.net/moodlebox-4.5.0-dev.img.bz2

Thank you for your help!

XECDesign, 2022-11-09:
I didn't see any pre-set password and the user config seemed to work. Don't know whether it interferes with moodle itself.

At a glance, cmdline.txt seems to point to the old init script. regenerate_ssh_host_keys.service is disabled, resolv.conf assumes a local router and IP for the nameserver (so nothing gets resolved when chrooted in), hosts file doesn't get updated properly when the hostname changes. I haven't checked, but I assume rpi-imager's wlan won't work and isn't expected to.

Distributing an image that has been booted into has all sorts of pitfalls. I think ansible has a way of operating on a chroot instead of SSH, so maybe that could be a cleaner approach, if pi-gen isn't suitable.

martignoni, 2022-11-09:
@XECDesign Thanks for your feedback. Some comments/questions:

> I didn't see any pre-set password and the user config seemed to work.

Thank you for the confirmation, I've indeed reset the change user routine at the end of the build process. FYI, this doesn't interfere with Moodle.

> cmdline.txt seems to point to the old init script

Is this a problem? (I try to revert to the original `cmdline.txt` at the end of the build process.)

> regenerate_ssh_host_keys.service is disabled 

Is this normal or should I change something? (I've disabled SSH at the end of the build process.)

> hosts file doesn't get updated properly when the hostname changes

This will be fixed, see https://github.com/moodlebox/moodlebox/issues/272.

> I haven't checked, but I assume rpi-imager's wlan won't work and isn't expected to

Right.

> Distributing an image that has been booted into has all sorts of pitfalls.

Could you elaborate which pitfalls? The problem is that several steps are still needed after the end of the build.

Thanks again for your time.

XECDesign, 2022-11-10:
> Is this a problem? (I try to revert to the original `cmdline.txt` at the end of the build process.)

It's not maintained and will be removed at some point. Note that the original `cmdline.txt` does not point to that file either.

> Is this normal or should I change something? (I've disabled SSH at the end of the build process.)

The new init script re-generates the ssh host keys, but if something goes wrong, `regenerate_ssh_host_keys.service` is a fallback. By using the old init script and disabling the service, you never end up re-generating the keys, leaving the user vulnerable to MITM attacks. Even if SSH is disabled by default, the image shouldn't contain any host keys. Generating them on first boot ensures they are there if the user happens to enable SSH.

> Could you elaborate which pitfalls?

Wouldn't be as much of a problem if they were all known. The general problem is that various applications and services like to generate unique IDs and keys, then when those unique strings are copied across multiple devices, they can misbehave. Some examples might be SSH host keys, the disk ID, systemd's machine ID, dhcpcd's DUID. You also end up with various cache files eating up space, log files and bash history, which might not be an issue, but could leak unintended information or may just be confusing to the user ("I don't remember running this, have I been hacked!?").

martignoni, 2022-11-10:
> Note that the original `cmdline.txt` does not point to that file either.

I've to fix that, thanks for the heads up.

> Some examples might be SSH host keys, the disk ID, systemd's machine ID, dhcpcd's DUID. You also end up with various cache files eating up space, log files and bash history, which might not be an [...]

Log files and shell history should not be a problem, since I remove them at the end of the build process. I did not delete them for the above "draft" image for lack of time.

Thanks again. With those fixes (hosts file, new init script), do you think MoodleBox has a chance to be added to default choices of RPi Imager?

XECDesign, 2022-11-10:
Yeah, the only real issue was having the system wide open by default. Otherwise, I don't see a problem.

martignoni, 2022-11-18:
@ghollingworth [This development image](https://moodlebox.net/moodlebox-4.5.0-dev-7.img.bz2) fixes all the issues raised by @XECDesign. I've tested it and it works correctly with RPi Imager.

What's the next step?

maxnet, 2022-11-20:
Note that your os_list.json needs to have a "extract_sha256" with the sha256 of the image before compression (so of the .img, not of the .img.bz2)

Like some others (not sure who started this trend), you do have a "image_download_sha256", but that property is actually not something supported by Imager.

martignoni, 2022-11-20:
> Note that your os_list.json needs to have a "extract_sha256" with the sha256 of the image before compression

Thank you. I've fixed this.

XECDesign, 2022-11-23:
> What's the next step?

Whenever you're happy with releasing it, update the json file on your end and send the snippet you would like us to include in [ours](https://downloads.raspberrypi.org/os_list_imagingutility_v3.json).

After that, I'll give everything a another quick glance over and add it.

martignoni, 2022-11-30:
The new MoodleBox image was released yesterday, and json file was updated accordingly.

Here's the snippet to include (suggestion: in "Other specific-purpose OS" if possible):
```json
{
    "name": "MoodleBox",
    "description": "MoodleBox is an open source distribution combining a wireless access point with a full featured Moodle server.",
    "website": "https://moodlebox.net",
    "icon": "https://moodlebox.net/img/icon.png",
    "subitems_url": "https://moodlebox.net/moodlebox-rpi-imager.json"
}
```

XECDesign, 2022-12-09:
Sorry about the delay. Should be up now.

martignoni, 2022-12-24:
Works perfectly, many thanks, and a merry Christmas 🎄

closed 2022-12-24 by martignoni

mudge, 2023-04-18:
Hi @martignoni, we're currently having trouble parsing the HTTP response from https://moodlebox.net/moodlebox-rpi-imager.json as it looks like you have a [`Content-Security-Policy` header](https://developer.mozilla.org/en-US/docs/Web/HTTP/Headers/Content-Security-Policy) with newlines in it (see the below output from `curl`) which is causing several of our HTTP clients to fail.

```
< HTTP/1.1 200 OK
< Server: nginx
< Date: Tue, 18 Apr 2023 09:18:19 GMT
< Content-Type: application/json
< Content-Length: 678
< Last-Modified: Mon, 20 Mar 2023 15:58:11 GMT
< Connection: keep-alive
< Vary: Accept-Encoding
< ETag: "64188293-2a6"
< Content-Security-Policy: 
< 		default-src 'self';
< 		script-src 
< 			'self'
< 			'unsafe-inline'
< 			https://stats.martignoni.net
< 			https://www.paypal.com
< 			https://cdn.jsdelivr.net;
< 		object-src 'none';
< 		style-src
< 			'self'
< 			'unsafe-inline'
< 			https://cdn.jsdelivr.net;
< 		img-src 'self' data: https://stats.martignoni.net;
< 		font-src 'self';
< 		connect-src 'self' https://stats.martignoni.net https://hooks.zapier.com;
< 		manifest-src 'self';
< 		media-src 'self';
< 		frame-src 'self' https://www.youtube.com https://www.youtube-nocookie.com https://object.center;
< 		frame-ancestors 'self';
< 		base-uri 'self';
< 		block-all-mixed-content;
< 	
< Strict-Transport-Security: max-age=31536000
< Referrer-Policy: strict-origin-when-cross-origin
< X-Frame-Options: SAMEORIGIN
< X-Content-Type-Options: nosniff
< X-XSS-Protection: 1; mode=block
< Permissions-Policy: interest-cohort=()
< Accept-Ranges: bytes
```

Would it be possible to update your server configuration to remove the newlines from your CSP header? The following should be a valid alternative:

```
Content-Security-Policy: default-src 'self'; script-src 'self' 'unsafe-inline' https://stats.martignoni.net https://www.paypal.com https://cdn.jsdelivr.net; object-src 'none'; style-src 'self' 'unsafe-inline' https://cdn.jsdelivr.net; img-src 'self' data: https://stats.martignoni.net; font-src 'self'; connect-src 'self' https://stats.martignoni.net https://hooks.zapier.com; manifest-src 'self'; media-src 'self'; frame-src 'self' https://www.youtube.com https://www.youtube-nocookie.com https://object.center; frame-ancestors 'self'; base-uri 'self'; block-all-mixed-content;
```

martignoni, 2023-04-18:
Out of curiosity, what's your failing client? (Since newlines are part of the spec, see https://w3c.github.io/webappsec-csp/#grammardef-serialized-policy and https://infra.spec.whatwg.org/#ascii-whitespace).

mudge, 2023-04-18:
It fails with [Ruby's `Net::HTTP`](https://docs.ruby-lang.org/en/3.2/Net/HTTP.html):

```ruby
# With default compression support enabled
Net::HTTP.get(URI('https://moodlebox.net/moodlebox-rpi-imager.json'))
Net::HTTPBadResponse: wrong chunk size line: ng: gzip
from /Users/mudge/.rubies/ruby-3.2.2/lib/ruby/3.2.0/net/http/response.rb:607:in `read_chunked'
```

```ruby
# With compression support explicitly disabled (note the headers included in the response body)
Net::HTTP.get(URI('https://moodlebox.net/moodlebox-rpi-imager.json'), 'Accept-Encoding' => 'identity')
=> "Strict-Transport-Security: max-age=31536000\r\nReferrer-Policy: strict-origin-when-cross-origin\r\nX-Frame-Options: SAMEORIGIN\r\nX-Content-Type-Options: nosniff\r\nX-XSS-Protection: 1; mode=block\r\nPermissions-Policy: interest-cohort=()\r\nAccept-Ranges: bytes\r\n\r\n{\n  \"os_list\": [\n    {\n      \"name\": \"MoodleBox v4.5.1\",\n      \"description\": \"MoodleBox combines a wireless access point with a full featured Moodle server on a Raspberry Pi. Compatible with RPi 4B, 400, 3B+, 3B and 3A+.\",\n      \"website\": \"https://moodlebox.net/\",\n      \"icon\": \"https://moodlebox.net/img/icon.png\",\n      \"url\": \"https://github.com/moodlebox/moodlebox/releases/download/v4.5.1/moodlebox-4.5.1-rpi-imager."
```

It also fails with the [HTTPie command-line client](https://httpie.io):

```console
$ http https://moodlebox.net/moodlebox-rpi-imager.json
HTTP/1.1 200 OK
	
		base-uri 'self';
		block-all-mixed-content;
		connect-src 'self' https://stats.martignoni.net https://hooks.zapier.com;
		default-src 'self';
		font-src 'self';
		frame-ancestors 'self';
		frame-src 'self' https://www.youtube.com https://www.youtube-nocookie.com https://object.center;
		img-src 'self' data: https://stats.martignoni.net;
		manifest-src 'self';
		media-src 'self';
		object-src 'none';
		script-src 'self' 'unsafe-inline' https://stats.martignoni.net https://www.paypal.com https://cdn.jsdelivr.net;
		style-src 'self' 'unsafe-inline' https://cdn.jsdelivr.net;
Connection: keep-alive
Content-Encoding: gzip
Content-Security-Policy: 
Content-Type: application/json
Date: Tue, 18 Apr 2023 09:39:43 GMT
ETag: W/"64188293-2a6"
Last-Modified: Mon, 20 Mar 2023 15:58:11 GMT
Permissions-Policy: interest-cohort=()
Referrer-Policy: strict-origin-when-cross-origin
Server: nginx
Strict-Transport-Security: max-age=31536000
Transfer-Encoding: chunked
Vary: Accept-Encoding
X-Content-Type-Options: nosniff
X-Frame-Options: SAMEORIGIN
X-XSS-Protection: 1; mode=block

{
    "os_list": [
        {
            "description": "MoodleBox combines a wireless access point with a full featured Moodle server on a Raspberry Pi. Compatible with RPi 4B, 400, 3B+, 3B and 3A+.",
            "extract_sha256": "ab2489742d1ac3ba9536a141dbfbac3495115045a4ea94d8032d271fcad26ae9",
            "extract_size": 2970615808,
            "icon": "https://moodlebox.net/img/icon.png",
            "image_download_size": 586110332,
            "init_format": "systemd",
            "name": "MoodleBox v4.5.1",
            "release_date": "2022-12-30",
            "url": "https://github.com/moodlebox/moodlebox/releases/download/v4.5.1/moodlebox-4.5.1-rpi-imager.img.xz",
            "website": "https://moodlebox.net/"
        }
    ]
}
```

mudge, 2023-04-18:
As you're returning a HTTP/1.1 response, it may be causing issues because [new lines were deprecated in RFC 7230](https://www.rfc-editor.org/rfc/rfc7230#section-3.2.4):

> Historically, HTTP header field values could be extended over multiple lines by preceding each extra line with at least one space or horizontal tab (obs-fold).  This specification deprecates such line folding except within the message/http media type ([Section 8.3.1](https://www.rfc-editor.org/rfc/rfc7230#section-8.3.1)).  A sender MUST NOT generate a message that includes line folding (i.e., that has any field-value that contains a match to the obs-fold rule) unless the message is intended for packaging within the message/http media type.

See also https://www.rfc-editor.org/rfc/rfc9110#section-5.5-5:

> Field values containing CR, LF, or NUL characters are invalid and dangerous, due to the varying ways that implementations might parse and interpret those characters; a recipient of CR, LF, or NUL within a field value MUST either reject the message or replace each of those characters with SP before further processing or forwarding of that message.

martignoni, 2023-04-18:
Done, but you'd better file an issue with Ruby and HTTPIE, because such a content policy with newlines is not uncommon.