raspberrypi/rpi-imager #500: Feature request

roontoon, 2022-11-04:
I would love to see a SD cloning feature in RPI-Imager. Thanks

lurch, 2022-11-04:
There's the "SD card copier" in Raspberry Pi OS, which _might_ do what you want? https://www.raspberrypi.com/news/another-update-raspbian/

roontoon, 2022-11-04:
Needs to run on a Mac. But it would still be a great feature for the app. Cheers.

> On Nov 4, 2022, at 1:39 PM, Andrew Scheller ***@***.***> wrote:
> 
> There's the "SD card copier" in Raspberry Pi OS, which might do what you want? https://www.raspberrypi.com/news/another-update-raspbian/
> 
>

tdewey-rpi, 2023-10-19:
I'm sorry, roontoon, but we don't have plans to introduce a direct SD card copying feature. If you're looking to image multiple cards for Raspberry Pi devices, you could issue multiple writes via the Imager.

Closing, as this isn't something we're looking at.

closed 2023-10-19 by tdewey-rpi