raspberrypi/rpi-imager #501: Error running authopen in MacOS Ventura 13.0

txoof, 2022-11-05:
**Hardware**

* Mac Mini M1 2020
* SanDisk Extrme Pro in generic external card reader

**Software**

* MacOS ventura 13.0
* Raspberry Pi Imager v1.7.3

**Issue**
Attempting to write an image using Raspberry Pi Imager results in error message: `Error running authopen to gain access to disk device '/dev/rdisk6' Please verify if 'Raspberry Pi Imager' is allowed access to 'removalble volumes' in privacy settings (under 'files and folders' or alternatively give it 'full disk access').` 

**Steps to recreate**
1. Open Raspberry Pi Imager
2. OS: Select Raspberry Pi OS Lite (64-Bit)
3. Storage: Select SanDisk sd card
5. Click Write
6. Enter admin password in popup
7. Error message appears immediately and opens System Preferences apps > * Privacy & Security > Files and Folders*

**Resolution Attempts**

* Set Files and Folders access for Raspberry Pi Imager, restart application
* Set Full Disk Access by clicking *Privacy & Security > Full Disk Access > +* and adding Raspberry Pi Imager explicitly; restart application
* Verify that slider is set to "ON" position for application

[image]

txoof, 2022-11-05:
<hangs head in shame> 100% User Error.

I was choosing a mounted external SSD rather than the card reader.

closed 2022-11-05 by txoof

maxnet, 2022-11-07:
>I was choosing a mounted external SSD rather than the card reader.

While it's good that it saved your data in this case, you were actually saved by a bug, rather than a feature.
It certainly do should be able overwrite any external USB drive as well, as some use those as storage for their Pi instead of SD cards.

Will get the bug that fails to unmount APFS formatted volumes (no matter if they are USB disks or SD cards) sorted.

txoof, 2022-11-08:
@maxnet 
When I realized how dumb I had been and how lucky I was, my stomach nearly fell out.

It would have been all kinds of tragic if I had wiped that SSD....

commit 30225187 (2022-11-07): OSX: fix unmounting drives that have APFS volumes

- If a drive is formatted APFS it will have a seperate disk
devices for physical drive (e.g. /dev/disk2) and volumes
(e.g. /dev/disk3).
Need to unmount all, or opening the device for
writing will subsequently fail.
(User will see an "Error running authopen" error in Imager
in that case).

- Also do not show APFS volumes seperately in the disk
selection dialog. List mount points under physical drive
instead.

Closes #501