raspberrypi/rpi-imager #502: Display actual status instead of just showing "Writing..."

NickyNiclas, 2022-11-09:
When downloading an image it's very unclear what is going on and can cause confusion, especially if the download happens to be really slow. It would be good if it could show the status "Downloading image" and the download speed to avoid confusion about why it's sometimes very slow.

maxnet, 2022-11-09:
It is a bit tricky.
We already start writing the image while it downloads, and will slow down the download if your SD card cannot keep it with the phase.

If we display download speed and it appears low, people may complain about "our slow servers", while that may not be the actual problem.

NickyNiclas, 2022-11-09:
Yeah, I was writing Ubuntu Server 22.04 LTS to an SSD via USB today and wondered why it was slow. It was the canonical server being slow but at first, I was wondering if something was wrong with my SSD as it took way longer than usual. Can't say it's critical but something to help avoid confusion would be great!

lurch, 2022-11-09:
I guess the only way around that would be to show "download speed" and "card write speed" as different figures; but that'd likely make the UX too confusing for people.

maxnet, 2022-11-09:
>I guess the only way around that would be to show "download speed" and "card write speed" as different figures

That will not make it clear to people which is affecting which.
If card is not responding, both download speed and card write speed will show 0.

We download at a maximum of card write speed.

(Theoretically the code could be refactored to download faster if it is caching the download to normal disk.
But the code currently does not do that, as cache file support was added later as an afterthought.).

MatthewCKelly, 2024-02-04:
perhaps a few updates are required..
1. update to "Downloading and writing" - no need to include speed.
2. Allow the end user to pre-download the files. And allow the users to manage the location of the cache files, rather than the users profile (Windows users for example).
I have a gb internet and the download should be quick...  I know the next build will be quicker as the content is cached..

lurch, 2024-02-05:
> And allow the users to manage the location of the cache files

See e.g. #300

closed 2025-12-12 by tdewey-rpi