raspberrypi/rpi-imager #503: Feature request: enable uart?

jedahan, 2022-11-13:
Would be nice in the config wheel to enable serial if we don't have internet connection

lurch, 2022-11-14:
See #373

timg236, 2022-11-14:
I think that PR is asking for uart_2ndstage=1 which is for firmware debug output. Adding "enable_uart=1" for a UART console would be useful. Although, I admit I am one of the probably small group of people who mostly connect to a Pi via a UART.

timg236, 2022-11-14:
btw: If you want a Pi 4 to ALWAYS have as UART console then add the following to the EEPROM config

```
# Always append config.txt with these device-configuration options
[config.txt]
[all]
enable_uart=1
```

tdewey-rpi, 2025-12-12:
Implemented in Imager, but requires turning on in the JSON on a per-device-and-OS basis: https://github.com/raspberrypi/rpi-imager/blob/f0054e8fd4739b1c2ba3f4d9ebb00586772f3392/doc/json-schema/os-list-schema.json#L266

Critically, custom image users can use that _today_.

closed 2025-12-12 by tdewey-rpi