raspberrypi/rpi-imager #505: Operating system did not mount fat32 partition

Pitorru, 2022-11-18:
Raspberry Pi OS Lite 64-bit
Storage to flash is a Seagate HDD 3.0USB
I'm using Ubuntu 22.04.1 LTS.

Play sound when finished UNCHECKED
Eject media when finished UNCHECKED
Enable telemetry CHECKED.

On Verify Step at the end tells me:

Operating system did not mount fat32 partition .

Help Please

maxnet, 2022-11-18:
>I'm using Ubuntu 22.04.1 LTS.

Using the .deb or a snap?

Can you try if this beta version works better?

[rpi-imager_1.7.4_amd64.deb](https://www.maxnet.eu/tmp/174-fat/rpi-imager_1.7.4_amd64.deb)

(No longer relies on operating system to mount FAT32 partition, but can modify it directly).

Pitorru, 2022-11-18:
> >I'm using Ubuntu 22.04.1 LTS.
> 
> Using the .deb or a snap?
> 
> 
> Can you try if this beta version works better?
> 
> [rpi-imager_1.7.4_amd64.deb](https://www.maxnet.eu/tmp/174-fat/rpi-imager_1.7.4_amd64.deb)
> 
> (No longer relies on operating system to mount FAT32 partition, but can modify it directly).

I'm using .deb, I Will try this beta and I Will respond you

Pitorru, 2022-11-18:
Solved, but when now I try to boot from USB it gives me some USB-MSD MSD [03:00] 3.00 000000:03 error 5

maxnet, 2022-11-18:
Do you also get the same bootloader error when not setting any "advanced settings"?
Do you also have a normal SD card to test with, instead of the USB HDD?

Hard to tell if the bootloader does not like your USB HDD (or if it does not have enough power), or if there is a bug in the new FAT32 modification code for the advanced settings.

Pitorru, 2022-11-18:
[image]

maxnet, 2022-11-18:
@timg236 is it correct that the RPI bootloader does not support USB hard drives that are using 4k sectors (instead of the traditional 512 bytes)?

rasulkireev, 2022-12-08:
I have the same issue. Using Mac rpi-imager, version 1.7.3. Installing Ubuntu Server 22.04 64bit on a 32GB SD Card. Tried both with and without Advanced Settings.

timg236, 2022-12-08:
Correct neither bootloader nor firmware support native 4K sectors

maxnet, 2022-12-08:
>Correct neither bootloader nor firmware support native 4K sectors

Ok, not something that can be solved in Imager then.

If there are no plans to add suport, you may want to consider adding a more clear error message though.
As you can see in this issue, users tend to look only at the last line only and "error 5" is not that clear.

====

>I have the same issue. Using Mac rpi-imager, version 1.7.3. Installing Ubuntu Server 22.04 64bit on a 32GB SD Card.

In your case (not 4k sectors, but a "operating did not mount fat32 partition" message), you may want to try if this Imager version works better: [Raspberry Pi Imager 1.7.4.dmg](https://www.maxnet.eu/tmp/174-fat/Raspberry%20Pi%20Imager%201.7.4.dmg)

markholland83, 2023-01-15:
> "operating did not mount fat32 partition"

[Raspberry Pi Imager 1.7.4.dmg](https://www.maxnet.eu/tmp/174-fat/Raspberry%20Pi%20Imager%201.7.4.dmg) resolved my issues, many thanks!

Xykrotik, 2023-01-17:
Same thing happening to me, Windows 10 64bit
Running 1.7.3 Raspberry Pi Imager

Attempting to install Raspberry Pi OS 64bit (Lite) to a USB external SSD (Samsung 960 Evo)

Also receiving "Operating system did not mount fat32 partition." Successfully erased the drive via RPi Imager and had it format to FAT32. Did not fix the issue

Unsure where to find 1.7.4beta of RPi Imager for windows, would appreciate a download link!

chi0tzp, 2023-02-09:
Similarly to what @rasulkireev reported, the problem disappeared when I tried **without modifying the advanced settings**.

lurch, 2023-02-09:
> Unsure where to find 1.7.4beta of RPi Imager for windows, would appreciate a download link!

There's a link in #542

mariuszbrz, 2023-09-09:
I had the same problem on linux system, so i tried the same version of "Raspberry PI Imager v1.7.3" with the same card reader, on the same usb port with windows - it worked. So i decide to dig a little bit deeper.
It appeared, that usb autosuspend caused the problem on linux.

If you use linux, type-in in terminal:
`cat /sys/module/usbcore/parameters/autosuspend`
if you get
`2`
It means you have autosuspend enabled.
type:
`echo '-1' | sudo tee /sys/module/usbcore/parameters/autosuspend`
to disable auto-suspend and retry write image to SD card.

To make change permanent, edit
`/etc/default/grub`
and add
`usbcore.autosuspend=-1`
to the end of the line starting with:
`GRUB_CMDLINE_LINUX_DEFAULT`
write this file and update grub config by typing:
`sudo update-grub`

In my case disabling usb auto-suspend fixed the problem with usb write on linux.
source: https://www.kernel.org/doc/Documentation/usb/power-management.txt

lurch, 2023-09-11:
From what little I know about USB, I think the USB device itself says whether it supports suspend or not? So perhaps your USB card reader is advertising that it's "suspendable", when it isn't? Or maybe there's a bug in the Linux kernel when it tries to "wake up" your card reader after it's been suspended?
My Linux laptop also has `2` set in `/sys/module/usbcore/parameters/autosuspend`, but I've never had any problems writing SD card images :man_shrugging:

IanEdington, 2023-09-12:
I didn't have this problem using an sd but did when I tried to use an SSD. I followed the steps above and they worked. Thank you @mariuszbrz 

I did have to unplug and re-plug my external sdd for it to take affect

crankyoldbugger, 2023-10-11:
Upgrading to the .deb version 1.7.5 seemed to fix the fat32 mount problem for me.

jdeltoft, 2023-11-28:
> > "operating did not mount fat32 partition"
> 
> [Raspberry Pi Imager 1.7.4.dmg](https://www.maxnet.eu/tmp/174-fat/Raspberry%20Pi%20Imager%201.7.4.dmg) resolved my issues, many thanks!

I've tried 1.7.4 on a 2019 macbook and 1.8.1 on a 2023 mac mini and both still produce this FAT32 error message after writing but I'm also always (so far) using advanced options. I can try without that but is there a workaround then to not having those advancced options set?

jdeltoft, 2023-11-28:
> Similarly to what @rasulkireev reported, the problem disappeared when I tried **without modifying the advanced settings**.

I'm wondering if the reports here of things working with .deb or 1.7.4 are really just this same resolution.  I.e. they see it working because they are not setting advanced options.

maxnet, 2023-11-28:
Where did you get the image you are writing?

Newer Imager versions are not capable of displaying that exact error message, unless you are writing a .zip file with multiple loose files in it.

shaakunthala, 2024-04-29:
I believe elevated privileges "expire" before rpi-imager completes the process. I can't think of any other hypothesis because whenever I run rpi-imager as root (sudo rpi-imager on Terminal), it works perfectly.

OlivierBouchoms, 2024-05-04:
> `cat /sys/module/usbcore/parameters/autosuspend`

> I didn't have this problem using an sd but did when I tried to use an SSD. I followed the steps above and they worked (with one addition). Thank you @mariuszbrz
> 
> I did have to unplug and re-plug my external sdd for it to work

Thanks, this also worked for me.

mariuszbrz, 2024-05-10:
> Wysłane: sobota, 4 maja 2024 14:33
> Do: raspberrypi/rpi-imager ***@***.***>
> DW: Brzezik Mariusz ***@***.***>; Mention ***@***.***>
> Temat: [Newsletter] Re: [raspberrypi/rpi-imager] Operating system did not mount fat32 partition (Issue #505)
>
> Bądź ostrożny: Wiadomość pochodzi z domeny niezarządzanej przez LUX MED. Nie otwieraj załączników ani nie klikaj w załączone odnośniki, jeśli nie rozpoznajesz nadawcy.

I'm very glad this helped you too. 🙂

[cid:apla_865088ea-e81f-4745-a1db-43d60e5ba5f8.png]

[cid:logo_e2197505-15a5-4471-9d23-ae0bcc23aefb.png]

LUX MED sp. z o.o. z siedzibą w Warszawie przy ul. Postępu 21C,
02-676 Warszawa, wpisana do rejestru przedsiębiorców prowadzonego przez Sąd Rejonowy dla m.st. Warszawy, XIII Wydział Gospodarczy Krajowego Rejestru Sądowego pod numerem KRS 0000265353, NIP 527-252-30-80 wysokość kapitału zakładowego: 676 123 500,00 zł. Więcej informacji o LUX MED Sp. z o.o. na stronie www.luxmed.pl

Treść tej wiadomości może zawierać treści poufne i jest przeznaczona wyłącznie dla jej adresata. Jeżeli nie jesteście Państwo jej adresatem, prosimy o powiadomienie o tym nadawcy wiadomości oraz jej usunięcie. Przetwarzamy Państwa dane osobowe. Szczegółowe informacje znajdziecie Państwo tutaj<https://www.luxmed.pl/assets/media/Przetwarzanie-danych-osobowych-w-zwiazku-z-prowadzona-wspolpraca-biznesowa.pdf>
This message and any attachments are confidential as a business secret and are intended solely for the use of the individual or entity to whom they are addressed. If you are not the intended recipient, please remove this message and contact with the sender. Click here<https://www.luxmed.pl/assets/media/Przetwarzanie-danych-osobowych-w-zwiazku-z-prowadzona-wspolpraca-biznesowa.pdf> to get detailed information about the processing your personal data.

dpfrakes, 2024-08-18:
> Runned into same issue with `Ubuntu 22.04 LTS` today Upgrading to rpi-imager `1.8.5` solved the issue.
> 
> Steps I did
> 
> 1. Download `1.7.5` release manually from Github at https://github.com/raspberrypi/rpi-imager/releases/tag/v1.8.5
>    Since `1.7.2` only is available for current Ubuntu LTS version : https://pkgs.org/download/rpi-imager
> 
> ```
> $ cat /etc/os-release | grep PRETTY_NAME
> PRETTY_NAME="Ubuntu 22.04.3 LTS"
> ```
> 
> 2. Remove former version and install `1.8.5` version
> 
> ```
> sudo apt remove rpi-imager
> cd /path/to/download
> sudo apt install ./rpi-imager_1.8.5_amd64.deb # https://unix.stackexchange.com/questions/159094/how-to-install-a-deb-file-by-dpkg-i-or-by-apt
> ```
> 
> Tested with moode audio custom image donwloaded from: https://moodeaudio.org/
> 
> Maybe this can help

Upgrading Imager worked for me as well, thanks for the tip!

Bandicoot, 2024-09-07:
I had the same problem with the default 1.7.2 included with Ubuntu 22.04 LTS.  Upgrading to 1.8.5 fixed the problem and let me image the CM4 boards I was having trouble with yesterday.  (Imaging w/o customization is not fun.)

tdewey-rpi, 2024-09-12:
Closing as fixed - looks like this got addressed in the 1.8.x series.

closed 2024-09-12 by tdewey-rpi