raspberrypi/rpi-imager #506: Unable to customize. File 'F:/config.txt' does not exist

Zilverdael80, 2022-11-21:
Trying to create a SD card to boot USB. 
Using version 1.7.3
Writing and Verifying goes with no issue it seems, 
[image]

Windows re-mounts the SD card as drive F: 
Imagers fails out, Unable to customize. File 'F:/config.txt' does not exist
[image]

File indeed is not on the drive, also not after re-mounting the drive in windows. 

I dont know what should be in this file, so basically i'm stuck now as i want to boot from USB and not SD.

maxnet, 2022-11-21:
Not sure why it is trying to apply advanced settings to the EEPROM image, which does not support that.

Did you write any other images before the EEPROM image in the same session?
If so, does it work better if you close Imager, and start it again, and write only the EEPROM image?

Zilverdael80, 2022-11-21:
That indeed fixed the issue, 
i wrote a home assistant OS image to USB drive and then started the SD write to boot from USB. 
Restarting the imager 'fixed' the issue

foghcz, 2023-02-26:
I'm having the same issue on MacOS

DanGlChris, 2023-03-25:
Yesterday, I had the same problem with different os files that I'd tried to boot on a sd card for a raspberry pi. 
I've solved it today just by unchecking all parameter here. And It works!!!

[image]

maxnet, 2023-03-25:
>Yesterday, I had the same problem with different os files that I'd tried to boot on a sd card for a raspberry pi.
>I've solved it today just by unchecking all parameter here. And It works!!!

If you do not customize any setting, it indeed has no need to alter the files on the FAT partition.
Alternatively, you may want to try a newer Imager version. 1.7.2 is old.

closed 2023-03-25 by maxnet