raspberrypi/rpi-imager #509: Wi-Fi Networks with parenthesis

Anthony6444, 2022-11-28:
After installing Rasberry Pi OS Lite (64bit) version 2022-09-22, with a wifi name that has a pair of parenthesis in it, `/boot/firstrun.sh`  fails with the following error:
```
[28.832781] firstrun.sh[276]: /boot/firstrun.sh: line 39: syntax error near unexpected token `)'
[28.834457] firstrun.sh[276]: /boot/firstrun.sh: line 39: `  /usr/lib/raspberrypi-sys-mods/imager_custom set_wlan 'Wifi (Taylor\'s Version)' '8d82....536' 'US'
```
[image]

Anthony6444, 2022-11-28:
reflashing with a different wifi name and booting again solves the issue, but I would really like to see this fixed!

maxnet, 2022-11-28:
Duplicate of #486 and already fixed in git.

closed 2022-11-28 by maxnet