raspberrypi/rpi-imager #511: CRDA is deprecated since Ubuntu Jammy / wifi location setting doesn't work anymore

BassMonkey, 2022-11-28:
Currently, rpi-imager sets the wifi **regulatory domain** with the following line in **user-data**:

```
runcmd:
- sed -i 's/^s*REGDOMAIN=S*/REGDOMAIN=DE/' /etc/default/crda || true
```

Since Ubuntu Jammy **crda is deprecated** and that approach doesn't work anymore. This is also being discussed here:
https://bugs.launchpad.net/netplan/+bug/1951586

As a result, **headless install with wifi is partially broken since Ubuntu 22.04**. Meaning, it won't connect to wifis on channels outside of what's allowed in the default reg domain 00.

In version 0.105 Netplan has added an option to set the regulatory domain for each wifi adapter. However, this is not yet part of any of the current images. It looks like the following and can be used either manually after upgrading Netplan via wifi or through cloud-init if the image already contains Netplan 0.105:

```
ubuntu@ubuntu:~$ cat /etc/netplan/50-cloud-init.yaml 
[...]
    wifis:
        wlan0:
            regulatory-domain: DE
            dhcp4: true
            optional: true
[...]
```

The only option that should already be working with current images is to set it directly in the kernel by adding `cfg80211.ieee80211_regdom=DE` (with the respective domain) to **cmdline.txt**.

maxnet, 2022-11-28:
>In version 0.105 Netplan has added an option to set the regulatory domain for each wifi adapter.

Yeah, we had added support for that, but had to revert it again.

https://github.com/raspberrypi/rpi-imager/commit/2c00c64aba9ed8052a47e39a6ae2d05e65a5a357

No official Ubuntu releases support that tag yet, and they error out the entire installation if it is encountered instead of just ignoring it. :-(

>The only option that should already be working with current images is to set it directly in the kernel by adding 
>cfg80211.ieee80211_regdom=DE (with the respective domain) to cmdline.txt.

This issue was reported to them over a year ago. (I submitted the bug report you linked)

Instead of us adding new stopgap solutions like that, I suggest Ubuntu backports support for the "regulatory-domain:" option to all Ubuntu image versions they offer in the repository instead.

@waveform80 what are your thoughts?

stek29, 2023-06-11:
Another alternative to avoid changing cmdline is to pass that parameter via modprobe.d, something like this:
```
$ cat /etc/modprobe.d/cfg80211_regdomain.conf 
options cfg80211 ieee80211_regdom=CONTRY-CODE
```

tdewey-rpi, 2025-12-12:
Fixed in 2.0+