raspberrypi/rpi-imager #513: Apple M1 macOS 12.6 corrupting CD cards

lantren, 2022-12-02:
I just attempted to used imager to create a new boot SD card with two micro SD cards and now both micro SD cards are locked and can't be erased on either windows or Mac M1.  Both cards were working previously as Pi boot SD's

There is no lock switch on the cards.  Attempted to remove write protect flags via software (diskutil on Mac and DISKPART on windows) to not avail. 

I was able to image another card on Windows successfully.   The 2 "locked" SD cards are now unusable

lantren, 2022-12-02:
the card was working before the attempt
Imager downloads assets and at  1% of the write it fails with an error. Verbatim unknown. I failed to write down the error dialog. 
Sadly there is no error logging I can look at and I don't have an expendable SD card at the moment. When I have another one to burn up , I'll try again and update with the exact error dialog

lantren, 2022-12-05:
I tried it again with fresh a SD card. I still couldn't write to it with imager , but this time it didn't trash the card.  

[image]

maxnet, 2022-12-05:
Is the card of the same brand as the ones that failed before?

Can also try launching Imager through: "Launchpad" -> "Other" -> "Terminal" and enter:

```
/Applications/Raspberry\ Pi\ Imager.app/Contents/MacOS/rpi-imager
```

Would be interested if it reports a more detailed write error through console.

As far as it "trashing" cards. If the write is interrupted you may end up with a card without partition table. That can be rectified by selecting "Erase, format as FAT32" as "operating system".

lantren, 2022-12-05:
One was the same, the other was not Actually, I did attempt to re-create the partition map on those cards on three different platforms, a Linux box a Windows machine, and my Mac m1.

They simple weren't recoverable.

miklosakos, 2022-12-25:
If you just want to burn an sd card, use an old 1.6.3 based Universal build, for images you're out of luck, none of the releases work properly as I've seen the whole slew of issues, including mine (#466). [Here's the link](https://github.com/raspberrypi/rpi-imager/issues/235#issuecomment-999791765) for the only universal build that I could get to work reliably.

lantren, 2022-12-26:
> If you just want to burn an sd card

Thank you very much for the information. I was really just reporting an issue I ran into. I was easily able to download the image file and burn the a new sd with balenaEtcher

dtgolder, 2025-12-07:
I have had the same issue--have bricked multiple SD cards on different laptops running Lubuntu. Tried on an older version running on a RPi and was able to salvage a bricked SSD. I believe this occurs when trying to Erase (format) an SD with multiple partitions (e.g, a disk formatted to install Linux Mint). It seems in such cases the "Erase" fails (forget the error message--sorry) and then the disk is no longer mountable. The only way I've found to recover that disk is to try installing a RPi image using RPi Imager (which can then somehow install a raspberry image) and then I was able to format (Erase) the drive using the OLD version of RPi imager on the Pi I had running. I'm now going to delete the new 2.0 image and go back to an earlier version so I no longer risk bricking any SD cards. (Note that once bricked I could not format in Windows or any other installer - I could only install a RPi image using the RPi installer). Unfortunately I've already tossed a few SD card before discovering this. Happy to assist with troubleshooting if necessary. Hopefully this description will help with a solution.

tdewey-rpi, 2025-12-12:
Closing this one as stale. 1372 caught a different issue.