raspberrypi/rpi-imager #515: Crashes after launching

RDKRACZ, 2022-12-04:
```
$ rpi-imager
QGLXContext: Failed to create dummy context
qrc:/main.qml:281:21: QML Rectangle: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
Failed to create OpenGL context for format QSurfaceFormat(version 2.0, options QFlags<QSurfaceFormat::FormatOption>(), depthBufferSize 24, redBufferSize -1, greenBufferSize -1, blueBufferSize -1, alphaBufferSize -1, stencilBufferSize 8, samples -1, swapBehavior QSurfaceFormat::DoubleBuffer, swapInterval 1, colorSpace QSurfaceFormat::DefaultColorSpace, profile  QSurfaceFormat::NoProfile) 
Aborted (core dumped)

```

maxnet, 2022-12-04:
>Failed to create OpenGL context

Does "which rpi-imager" show /usr/bin/rpi-imager or something else like "/snap/..." ?

What Linux distribution, GPU card?
You do are sitting in front of that computer and started "rpi-imager" locally from a local terminal window, and not through ssh or other remote means?

RDKRACZ, 2022-12-04:
>which rpi-imager

/usr/bin/rpi-imager

>What Linux distribution, GPU card

It's an old laptop with Mint 21, Geforce 920M.

>You do are sitting in front of that computer and started "rpi-imager" locally from a local terminal window, and not through ssh or other remote means?

Yes

maxnet, 2022-12-04:
Running other GPU accelerated programs does work?
E.g. "glxgears" (from "mesa-utils" package).

hnariman, 2024-08-14:
Is this issue still open?

also having this issue on Fedora 39 (workstation) - rpi-imager installed through dnf 

```shell
$ rpi-imager 
     
QGLXContext: Failed to create dummy context
Failed to create OpenGL context for format QSurfaceFormat(version 2.0, options QFlags<QSurfaceFormat::FormatOption>(), depthBufferSize 24, redBufferSize -1, greenBufferSize -1, blueBufferSize -1, alphaBufferSize -1, stencilBufferSize 8, samples -1, swapBehavior QSurfaceFormat::DoubleBuffer, swapInterval 1, colorSpace QSurfaceFormat::DefaultColorSpace, profile  QSurfaceFormat::NoProfile) 
[1]    687787 IOT instruction (core dumped)  rpi-imager
```

```shell
$ which rpi-imager
/usr/bin/rpi-imager
```

```shell 
lspci | rg VGA  
01:00.0 VGA compatible controller: NVIDIA Corporation GA104 [GeForce RTX 3070 Ti] (rev a1)
```

JulianAssmann, 2025-05-04:
I have the same issue.
```
$ which rpi-imager
/usr/bin/rpi-imager
```

```
$ rpi-imager
did not find extension DRI_IMAGE_DRIVER version 1
did not find extension DRI_Mesa version 1
failed to load driver: iris
did not find extension DRI_Mesa version 1
failed to load driver: iris
did not find extension DRI_Mesa version 1
QGLXContext: Failed to create dummy context
Failed to create OpenGL context for format QSurfaceFormat(version 2.0, options QFlags<QSurfaceFormat::FormatOption>(), depthBufferSize 24, redBufferSize -1, greenBufferSize -1, blueBufferSize -1, alphaBufferSize -1, stencilBufferSize 8, samples -1, swapBehavior QSurfaceFormat::DoubleBuffer, swapInterval 1, colorSpace QSurfaceFormat::DefaultColorSpace, profile  QSurfaceFormat::NoProfile) 
Aborted (core dumped)
```

This is the output regardless of whether I install it using `sudo apt-get rpi-imager` or by downloading and installing the `.dep` file.

closed 2025-12-12 by tdewey-rpi