raspberrypi/rpi-imager #516: Feature request: write final image into custom file or device

jimis, 2022-12-04:
I would like to download and prepare an image using `rpi-imager`, but **write it into a custom file**. 

The goal is to do further work on the image, before I copy it into one or several SD cards.

P.S. The help text shows an argument `<image file to write>` but this seems to serve a different purpose.
```
rpi-imager [--debug] [--version] [--repo <repository URL>] [--qm <custom qm translation file>] [--disable-telemetry] [<image file to write>]
-OR- rpi-imager --cli [--disable-verify] [--sha256 <expected hash>] [--debug] [--quiet] <image file to write> <destination drive device>
```

lurch, 2022-12-04:
...although I'm not sure why you need to use `rpi-imager` to do that?
```
$ wget -q -O- https://releases.libreelec.tv/LibreELEC-RPi4.arm-10.0.3.img.gz | gunzip > test.img
$ ls -l test.img
-rw-rw-r-- 1 andrew andrew 575668224 Dec  4 16:52 test.img
```

jimis, 2022-12-04:
Thanks! Unfortunately that doesn't seem to work in GUI mode.

 I want to use the GUI to do the basic setup (SSH, username, locale).

maxnet, 2022-12-04:
>Thanks! Unfortunately that doesn't seem to work in GUI mode.

Sorry, prefer not to allow writing arbitrary devices there.

>I want to use the GUI to do the basic setup (network setup, enable SSH, set username, set locale).

If you write a customized SD card once, and steal the firstrun.sh from the FAT partition, Imager v1.7.4 do can apply the same firstrun.sh to later images.

```
rpi-imager --cli --first-run-script firstrun.sh [image file/url] [device to write to]
```

(option was added for automated testing recently.
Imager v1.7.4 is not released yet, but you do can compile it from source, or grab a [.deb here](https://www.maxnet.eu/tmp/174-fat/rpi-imager_1.7.4_amd64.deb)).

jimis, 2022-12-07:
If the file `firstrun.sh` is all that is needed, would it be possible to only write that one somewhere?

maxnet, 2022-12-07:
>If the file firstrun.sh is all that is needed, would it be possible to only write that one somewhere?

If it is a RPI OS based OS you can manually toss on the FAT partition:

- firstrun.sh
- a modified cmdline.txt that has ` systemd.run=/boot/firstrun.sh systemd.run_success_action=reboot systemd.unit=kernel-command-line.target ` on the same line as existing options.

If it's a Ubuntu based OS you need the following files on the FAT partitioninstead:

- user-data
- network-config

jimis, 2022-12-20:
> If the file firstrun.sh is all that is needed, would it be possible to only write that one somewhere?

Is it possible for rpi-imager to write only this file somewhere, instead of writing the full image? This way I could customize this file and then manually add it to the installation image.

Thank you for the information, you have already provided very useful insight in what rpi-imager actually does.

maxnet, 2022-12-20:
>Is it possible for rpi-imager to write only this file somewhere, instead of writing the full image?

No.
Although it does print it to console for debugging purposes if you start "rpi-imager" from terminal and "save" the custom settings.

(Do need to replace \n with a real new line in the output if using that output)