raspberrypi/rpi-imager #517: macOS/Apple Silicon Universal binary?

tisba, 2022-12-04:
https://github.com/raspberrypi/rpi-imager/issues/235#issuecomment-999791765

ghollingworth, 2022-12-04:
Not until it becomes a little more prevalent.  Currently, only 0.2% of downloads come from an M1 Mac.

closed 2022-12-04 by ghollingworth

tisba, 2022-12-05:
Okay, I see the chicken and egg problem :) Especially if a project is not super straight forward to port. Out of curiosity: 0.2% of all mac downloads or overall?

@maxnet could you share how you build darwin-arm64 (or even the universal binary?). Having a native version seems to have helped quite a bit with issues users were having (skimming through the issues). Maybe a HOWTO can be put somewhere (README?) how to do it yourself.

maxnet, 2022-12-05:
>@maxnet could you share how you build darwin-arm64 (or even the universal binary?).

The main things:

- use Qt 6.3 instead of Qt 5
- search and replace "qt5" for "qt6" and link in "Qt::Core5Compat" in the CMakeLists.txt.
- add a CMAKE_OSX_ARCHITECTURES define (set(CMAKE_OSX_ARCHITECTURES "x86_64;arm64" CACHE STRING "" FORCE))

>Having a native version seems to have helped quite a bit with issues users were having (skimming through the issues).

I am not sure if those issues were really solved by the native version, or if that was more placebo effect.
People also reported issues with things like formatting as FAT32, while the role of Imager in that case is limited to starting OS X's diskutil to do the real work, and that does not run under Rosetta.

>Maybe a HOWTO can be put somewhere (README?) how to do it yourself.

Not sure if there is much demand for that.
Also be aware that if you want to be able to run the files you compiled afterwards on another Mac different than the one you compiled it on, you'll need a paid Apple developer subscription.

tisba, 2022-12-05:
hmm, not super straight forward 🤔 I don't see documentation how the release process works, otherwise I'd given it a shot to work on a PR...

I'm actually not really only asking for myself (I can do it without this/an UI tool). But I got asked a couple of times already by less tech savvy folks and they all bought new MacBook Airs in the last two years. While I'm very well aware of my bubble, but I do believe that the percentage of Mac users on arm64 will rapidly increase rather soon.

maxnet, 2022-12-05:
>Sorry, I meant, I was looking for an (automated) script/tool to build and publish releases.

I know that in the web application development world everyone is a fan of continuous integration and the "push to production" button.
But in the traditional application development world, in which new releases are only done a few times a year -instead of daily-, and the application needs to be code signed, and submitted to Apple for notarization first, build systems are typically still powered by human being. :-)

>I was hoping that there is something automated so it would be little to no effort offering builds for another build target.

It is a little bit more work, than a recompile.
As it uses different libraries.

Qt 5 not supporting M1.
And Qt 6 does support M1, but does not support older OSX versions.
It also uses a different graphic backend (Metal on Apple, and Vulkan on Linux, while it used OpenGL previously), so may have new bugs.

And that older OS releases are no longer supported is not just an Apple problem.
Qt 6 also drops support for Windows 7 and 8.
And older Linux distributions including RPI OS.
So we are not in a hurry to make that the new default.

tisba, 2022-12-05:
Thanks for the context and taking the time to elaborate! Much appreciated! 🙇

lurch, 2022-12-06:
> I know that in the web application development world everyone is a fan of continuous integration and the "push to production" button.

**cough** https://github.com/balena-io/etcher/issues/3853 **cough** :wink: 

I for one, appreciate the more traditional approach used by Raspberry Pi Imager :heart:

ttamg, 2022-12-13:
@tisba did you get an Apple Silicon built?  If so do you have it somewhere on GitHub that others can use it?  At their own risk of course!  Thanks!

ttamg, 2022-12-14:
Great @tisba .

A partial workaround for anyone hitting this thread and wanting to run an emulated Raspian on new apple silicon is to run it in a virtual machine on OSX

* Note that it doesn't work within VirtualBox on the Apple silicon
* It does work using the free UTM https://mac.getutm.app virtual machine within OS X as that allows emulation on a PC architecture
* It probably also works using Parallels, but as that is an expensive bit of software I don't have installed I have not tested that

Within UTM it's horribly slow so doesn't replace having an image that works on apple silicon, but is a workaround that I did get up and running yesterday.

infinitepower18, 2022-12-26:
> Not sure if there is much demand for that. Also be aware that if you want to be able to run the files you compiled afterwards on another Mac different than the one you compiled it on, you'll need a paid Apple developer subscription.
The paid subscription is only needed to get rid of the warning messages when you launch it the first time, otherwise you can get around that by right clicking + open.

tisba, 2022-12-27:
Without any documentation (or automation as documentation), it is - at least for me - unfeasible to make a build (any actually). The qt world seems super complex and extensive. I've seen a bunch of other qt-based projects which did automate their build and release pipeline, but I'm still lacking the experience to adopt this to this project within a reasonable time window 😞 

I thought I could help less savvy people here, but I just don't have the time or the leisure to spend days learning build processes without documentation. I can help myself with `dd` & friends and I hope that either this project or etcher figure out the arm+rosetta glitches for everybody who isn't comfortable with "lower-level" tools.

maxnet, 2022-12-27:
>Without any documentation (or automation as documentation), it is - at least for me - unfeasible to make a build (any actually).

That my documentation skills leave to be desired is correct :-)
Not being an English speaker writing anything in English takes me 10 times the time it would take someone else.

If you need help it may be useful if you be very precise indicating which part of the build process from the README you are having problems with.
E.g. if you do not understand which options you need to select in the Qt installer (actually think the defaults you get when you click "next" until "finish" will do) , or if you have problems adjusting the CMakelists for Qt 6.