raspberrypi/rpi-imager #521: Error 416 on 91.189.91.123

Theo-Dancoisne, 2022-12-15:
I don't know if you can do anything but i get the error 416 toward the end of the writing of Ubuntu desktop 22.10 64b.
[image]

maxnet, 2022-12-15:
HTTP error 416 = Range Not Satisfiable
Generally means download resuming -which we use if the connection to the download server is lost, or stalls-  does not work properly.

Pretty sure Ubuntu's servers do support that.
Any (corporate) security software / proxy servers / parental control software active?
Some break download resuming purposefully, as they cannot scan content properly if they do not see the start of the file because it is resuming from some later point.

Theo-Dancoisne, 2022-12-16:
No proxy or parental control software, and i tested on multiple networks.
Can you try in your side ?

maxnet, 2022-12-16:
>Can you try in your side ?

Works fine here.
If I provoke connection loss with iptables, it does a Range based resume fine.

```
$ rpi-imager
Fontconfig warning: "/etc/fonts/conf.avail/29-language-selector-ja-jp.conf", line 57: Having multiple values in <test> isn't supported and may not work as expected
Fontconfig warning: "/etc/fonts/conf.avail/65-ttf-droid-sans-fonts.conf", line 61: Having multiple values in <test> isn't supported and may not work as expected
Fontconfig warning: "/etc/fonts/conf.avail/65-ttf-droid-sans-fonts.conf", line 96: Having multiple values in <test> isn't supported and may not work as expected
qrc:/main.qml:281:21: QML Rectangle: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
Available disk space for caching: 438 GB
Unmounting: "/dev/sdc"
BLKDISCARD not supported
Zeroing out first and last MB of drive
Done zeroing out start and end of drive. Took 0 seconds
Image URL: "http://cdimage.ubuntu.com/releases/kinetic/release/ubuntu-22.10-preinstalled-desktop-arm64+raspi.img.xz"
Received header: "HTTP/1.1 200 OK"
Received header: "Date: Fri, 16 Dec 2022 13:35:36 GMT"
Received header: "Server: Apache/2.4.29 (Ubuntu)"
Received header: "Last-Modified: Tue, 18 Oct 2022 15:06:12 GMT"
Received header: "ETag: \"80fe2aec-5eb506ec70100\""
Received header: "Accept-Ranges: bytes"
Received header: "Content-Length: 2164140780"
Received header: "Content-Type: application/x-xz"
Received header: ""
HTTP connection lost. Time: 1671197811
Received header: "HTTP/1.1 206 Partial Content"
Received header: "Date: Fri, 16 Dec 2022 13:36:57 GMT"
Received header: "Server: Apache/2.4.29 (Ubuntu)"
Received header: "Last-Modified: Tue, 18 Oct 2022 15:06:12 GMT"
Received header: "ETag: \"80fe2aec-5eb506ec70100\""
Received header: "Accept-Ranges: bytes"
Received header: "Content-Length: 2017328284"
Received header: "Content-Range: bytes 146812496-2164140779/2164140780"
Received header: "Content-Type: application/x-xz"
Received header: ""
```

If you have a wired Ethernet connection, and a Pi with recent firmware, you can also run Imager on the Pi itself (remove SD card during boot, and press "shift" for network install), instead of on your normal computer.

Theo-Dancoisne, 2022-12-16:
So it's my fault.
Instead I downloaded the image myself on the Ubuntu website
https://ubuntu.com/download/raspberry-pi/thank-you?version=22.10&architecture=desktop-arm64+raspi
Thank you.

closed 2022-12-16 by Theo-Dancoisne