raspberrypi/rpi-imager #525: Error writing to storage

kolbai, 2022-12-21:
I always get this error. I have tried different SD cards, but result is the same. I am running Windows 10. Any suggestions how to fix this issue?

[image]

maxnet, 2022-12-21:
>I have tried different SD cards

Suggest you try a different SD card reader as well, if you have one.

KW7DSP, 2023-01-28:
I'm having the same problem.

KW7DSP@Hotmail.Com

Jasoniver, 2023-03-24:
I see this is problem that many have had before im trying to load Mainsail Os to my pi with a Rpi that was running octoprint and using the same SD 16gb sandisk SD card but having the error writing to disk

kalph208, 2023-07-01:
Same error. Tried with different sd card reader, tried multiple micro sd cards. Tried formatting fat32 on windows using built in formatter and easeUS Partition master. Says it successfully formats. Try to run raspberry pi imager, throws error, then computer will never read that card again.

aburacze, 2023-08-26:
Please have a look at https://www.reddit.com/r/rasberrypi/comments/13bmqsh/error_writing_to_storage/
Windows Defender may prevent Raspberry Pi imager accessing SD cards directly and you need to add rpi-imager.exe to "Controlled Folder Access".

maxnet, 2023-08-26:
>Windows Defender may prevent Raspberry Pi imager accessing SD cards directly and you need to add rpi-imager.exe 
>to "Controlled Folder Access".

Ah, yes.
Error reporting could be improved in regards to that.
We used to have a test for Controlled Folder Access, that displayed a clearer error message to the user if it may have failed due to that, but it's in a code path that is no longer used.

(Windows itself also does warn the user if it blocked an application, but their message is only there for a second in the top right corner of the screen before it dissapears to the "notifications" that not many people read).

Karlheinzniebuhr, 2023-09-13:
Same issue, its not the reader since windows has no problem accessing the card

Karlheinzniebuhr, 2023-09-13:
> Please have a look at https://www.reddit.com/r/rasberrypi/comments/13bmqsh/error_writing_to_storage/ Windows Defender may prevent Raspberry Pi imager accessing SD cards directly and you need to add rpi-imager.exe to "Controlled Folder Access".

this solved it for me thanks!