raspberrypi/rpi-imager #531: Unable to log into Pi

Firedogs2x, 2023-01-03:
I have a Pi 4B and I have reinstalled the latest OS 64 Full onto my SD card using the Imager 1.7.3.
I have used the advanced settings to set up wireless, SSH login information. I set a custom name and password.
I have verified that the Pi is connecting to my wireless network.
I use Putty to try and log in. After inputting the username, I enter the password and all I get back is access denied.
I was looking at one of your website pages "https://www.raspberrypi.com/news/raspberry-pi-bullseye-update-april-2022/". And on that page, it says to add a text file. But as you keep reading it says you need to encrypt your password by running a certain program on your Pi. Well, if I cannot log onto my Pi how am I supposed to run a program on it to encrypt my password to allow me to log on.
Please see attached document to see all of the advanced settings, see my SD card image, error message logging in, and the section I mentioned regarding setting up a text file for your log in info.

Thank you for all of your help.

[Can't Log into my Pi 01-03-2023.pdf](https://github.com/raspberrypi/rpi-imager/files/10335582/Can.t.Log.into.my.Pi.01-03-2023.pdf)

Xykrotik, 2023-01-17:
I was having an issue previously, similar but not exactly the same. 

My fix was instead of running `ssh pi@[IP ADDRESS]` which provided me with an error/inability to log into my Pi, 
I used `ssh pi@raspberrypi.local` and I was able to connect just fine. It's what I continue to use now, as it's been the most consistent with allowing me to successfully ssh into my pi

lurch, 2023-01-17:
> I was looking at one of your website pages "https://www.raspberrypi.com/news/raspberry-pi-bullseye-update-april-2022/". And on that page, it says to add a text file. But as you keep reading it says [...]

I think you missed the bit in that article where it says "There are also mechanisms to preconfigure an image without using Imager." i.e. if you _are_ using Imager, then you don't need to manually add any text files.

Have you double and triple checked that the username and password you entered in Imager are the same that you're using to try and log into your Pi?

Firedogs2x, 2023-01-17:
Hello, thank you for your comments.
Yes, it sounds like I did miss that information. It just made no sense to me when I read it, so I thought it was typed wrong.
Yes, I was 100% sure of my typing and use of the same username and password.
So, an update. I took an old laptop and wiped it clean, reinstalled windows 10 and the Pi imager.
Because one of my goals was to be able to run Octoprint. I found at Octoprint an image already setup with Pi OS and Octoprint.
It uses the Pi Imager to create the SD card.
It worked flawlessly absolutely no issues logging into the Pi using my created username and password. 
I plan to try creating an SD card with the OS x64 image and see if it works. If it does, then there is something on my main computer that is somehow causing issues when it comes to creating information that is specifically used for creating the custom user file.
Why it would only effect that specific part of the image I have no idea. But I am not a programmer, so I have no clue.
I just know by my searching the web this is not an isolated occurrence.

Again, Thank You very much for your help.

lurch, 2023-01-17:
> If it does, then there is something on my main computer that is somehow causing issues when it comes to creating information that is specifically used for creating the custom user file.
Why it would only effect that specific part of the image I have no idea.

Yes, seems like the "Advanced settings" sometimes don't work properly on some Windows computers e.g. #445

jenishngl, 2023-06-28:
I used the imager 1.7.5 today to write OS 64 Lite into my Raspberry Pi 4 and have used advanced settings to configure username/password & i have placed userconf file into the boot location as well. Still when i try to login, I am unable to login, its giving an error as "Permission denied, please try again."

lurch, 2023-06-28:
>  have used advanced settings to configure username/password & i have placed userconf file into the boot location as well.

If you're using Imager to configure your image, then there's no need to create a userconf file.

jenishngl, 2023-06-28:
I tried this like 15 times today again and again, still no luck.

I switched to using ssh authentication keys and only then I was able to
login into raspberry pi today.

Seems something is wrong with the imager app running in windows 11

--Jenish

maxnet, 2023-07-05:
>I tried this like 15 times today again and again, still no luck.

You do are using a normal user name, all lowercase, no spaces, and not being root?
And do not have any non-latin characters in your password?

jenishngl, 2023-07-05:
No, I don't have anything special with my password. Simple characters and
numbers. Username is just 'pi'

--Jenish

forrie, 2023-07-15:
Yeah, I totally missed this.  Would it make sense to have these advanced configuration options more obvious in  the UI?  For example, another button/tab "Configure Settings"?

peksonite, 2023-10-27:
Did you fix this issue? currently i've got same problem with mu pi zero 2 W. I've tried 10 times maybe, with 100% sure login and password, (no special characters, just 'pi' login and simple password with lowercase letters and number). And i cannot log in through ssh or simply by normal boot

oopiicaa, 2023-11-02:
No, it's not fixed. I'm having the same issue. There is a workaround described in [this](https://forums.raspberrypi.com/viewtopic.php?t=333248&sid=5ab6a3e0f4a1da0670bf9cdeef54bd1e#p1994926) post.

Firedogs2x, 2023-11-02:
Hello everyone.

The only way I was finally able to get my Pi up and running was to take an old laptop and do a fresh install of windows.
Then I installed only the Pi OS installer.
For some reason that worked.
I recently attempted to re-flash my OS from 32bit to the 64bit version. But it failed.
So, I re-flashed the 32bit version and everything worked fine.
I wish I had better news for all of you.
Best of luck.

jr1221, 2023-11-13:
I know this is dumb, but try your password without the first character.  I found a subtle UI glitch where your first character is used to clear a highlighted section and is not recorded.

fanyang-butlr, 2023-11-18:
> I know this is dumb, but try your password without the first character. I found a subtle UI glitch where your first character is used to clear a highlighted section and is not recorded.

This actually worked for me, thx!
As reference, I used Raspberry Pi imager v1.8.1 on Mac OS

njrusty, 2023-11-18:
I am in the same situation.  User/password are both pi, created in imager on a Win11 pc.  Was able to ssh in to my RP4 for a few time today with no issues.  Then all of a sudden, the RP4 will not accept my password.  So, was working with pi password and now will no longer accept it??????????

danswiser, 2023-12-01:
> I know this is dumb, but try your password without the first character. I found a subtle UI glitch where your first character is used to clear a highlighted section and is not recorded.

This was also the problem for me.

lurch, 2023-12-02:
@njrusty If it _was_ working and now it's not, then it's obviously not a problem with Raspberry Pi Imager! You should probably ask for help with your problem at https://forums.raspberrypi.com/

nul800sebastiaan, 2023-12-02:
This has been driving me crazy over the past few days, only on rare occasions would my password work. 

@jr1221's comment is SPOT ON! https://github.com/raspberrypi/rpi-imager/issues/531#issuecomment-1809201753 it works every time now, leaving off the first character of the password. Did 3 clean images, all 3 times did I need to leave off the first char of the password.

ZachreyNM, 2023-12-14:
> I know this is dumb, but try your password without the first character. I found a subtle UI glitch where your first character is used to clear a highlighted section and is not recorded.

HALLELUYAH!!! THANK YOU !!! THANK YOU!! THANK YOU!!

It worked!! 

I have been struggling many hours making images for Raspberry Pi Zero W. I downloaded the 32bit Lite version from raspberrypi dot com and used the rpi-imager on my Linux Manjaro laptop to flash an 8GB Sandisk uSD card for a Rpi Zero W. I selected rpi zero for the board, then I selected "custom" since I already downloaded the lite 32bit jobber, then I selected the 8GB card and clicked on next. This brought up the dialog box to set customizations but what was odd is that the boxes were already filled in (due to previous sessions with the tool). When I clicked on "show password" for the login password and the WiFi password, they were both these really long strings that looked encrypted. I "select all" and then type in the correct password, then enable SSH, etc and then started the flash.
The pi would boot up and apparently, connect to the network since I could SSH to it. But I could not log in! Even with a monitor connected to the RPiZW, I could not log in with username and password. 
I tried searching my entire laptop for the hostname I entered into the rpi-imager but could not find anything. How does rpi-imager know what I had entered previously? It must have created or edited a file somewhere on the disk, right?? If I could find that file, maybe I could clear it or delete it and the rpi-imager tool would work correctly??

tdewey-rpi, 2023-12-15:
Renaming this issue, now we have a reproduction mechanism I can look at it in earnest.

I can confirm I can replicate it on macOS Sonoma, but _only when editing a previously saved password_.

This appears to be an artefact of the code to prevent modifying a hashed password: 

```
                                            if (alreadyCrypted) {
                                                /* User is trying to edit saved
                                                   (crypted) password, clear field */
                                                alreadyCrypted = false
                                                clear()
                                            }
                                            if (indicateError) {
                                                indicateError = false
                                            }
```

In this case, I believe the first "keypress" gets put into the text field, triggers the onTextEdited() signal, which then calls clear(), destroying the input.

tdewey-rpi, 2023-12-15:
I think there's two things we can do to make this experience better:

1) Change the password masking to show the last character entered as you enter it
2) Move the clearance function to the key press handler chain

jr1221, 2023-12-15:
Thank you for figuring this out in a timely manner!

lurch, 2023-12-17:
> Change the password masking to show the last character entered as you enter it

That might be useful for touch-typists, but less useful for people who need to look at the keyboard when typing? :wink: 
Perhaps it'd be more consistent to just have a "Show password" tickbox, like there is for the WiFi password?

tdewey-rpi, 2023-12-17:
A fair point indeed, @lurch, and one I've pre-empted!

It turns out that PasswordEchoOnEntry will keep the entire password in the clear until you select another control in my testing (WSL2). I'll want to try it on a few more platforms yet to confirm it's doing what we expect, but I think this is the most useful behaviour - it avoids us storing the cleartext, has some straightforward behavioural semantics, and should allow for anyone to review their password before confirming it.

tdewey-rpi, 2023-12-17:
Regarding the WiFi password, I think we could make a case for changing that TextField to use the same behaviour, and drop the "show" button.

All subject to further testing, of course.

lurch, 2023-12-17:
> It turns out that PasswordEchoOnEntry will keep the entire password in the clear until you select another control

I guess the only slight wrinkle with that might be if somebody is using RPi Imager in a busy shared environment (e.g. in a school classroom) and they don't want somebody else to see what password they entered, just by watching their screen? :shrug:

tdewey-rpi, 2023-12-18:
This change will feature in v1.8.4, which I'm aiming to release this week.

closed 2023-12-18 by tdewey-rpi

Kovshische, 2024-01-21:
> This change will feature in v1.8.4, which I'm aiming to release this week.

@tdewey-rpi Look like the bug is still present in v1.8.4 on MacOS

tdewey-rpi, 2024-01-22:
> > This change will feature in v1.8.4, which I'm aiming to release this week.
> 
> @tdewey-rpi Look like the bug is still present in v1.8.4 on MacOS

Please raise a new issue for this, rather than commenting on a closed one.

FWIW - I've just tried v1.8.4 on Sonoma, and do not see this issue.

PR #753 (2023-12-18): qml: Clear saved password using onPressed handler
Using onTextEdited meant that the clear() call would happen after the edit had actually happened.

Instead, use the Keys.onPressed handler, which happens earlier in the input chain.

Resolves #531