raspberrypi/rpi-imager #533: Installing on windows subsystem for linux wsl unsuccesful using .deb

mhangaard, 2023-01-07:
System WSL with Ubuntu 20.04.5 LTS

Is it even possible to use rpi-imager on WSL without a desktop?
```
$ wget https://downloads.raspberrypi.org/imager/imager_latest_amd64.deb
```
```
$ sudo dpkg -i imager_latest_amd64.deb
Selecting previously unselected package rpi-imager.
(Reading database ... 44696 files and directories currently installed.)
Preparing to unpack imager_latest_amd64.deb ...
Unpacking rpi-imager (1.7.3) ...
dpkg: dependency problems prevent configuration of rpi-imager:
 rpi-imager depends on libqt5core5a (>= 5.12.2); however:
  Package libqt5core5a is not installed.
 rpi-imager depends on libqt5dbus5 (>= 5.0.2); however:
  Package libqt5dbus5 is not installed.
 rpi-imager depends on libqt5gui5 (>= 5.10.0) | libqt5gui5-gles (>= 5.10.0); however:
  Package libqt5gui5 is not installed.
  Package libqt5gui5-gles is not installed.
 rpi-imager depends on libqt5network5 (>= 5.12.2); however:
  Package libqt5network5 is not installed.
 rpi-imager depends on libqt5qml5 (>= 5.10.0); however:
  Package libqt5qml5 is not installed.
 rpi-imager depends on libqt5widgets5 (>= 5.0.2); however:
  Package libqt5widgets5 is not installed.
 rpi-imager depends on qml-module-qtquick2; however:
  Package qml-module-qtquick2 is not installed.
 rpi-imager depends on qml-module-qtquick-controls2; however:
  Package qml-module-qtquick-controls2 is not installed.
 rpi-imager depends on qml-module-qtquick-layouts; however:
  Package qml-module-qtquick-layouts is not installed.
 rpi-imager depends on qml-module-qtquick-templates2; however:
  Package qml-module-qtquick-templates2 is not installed.
 rpi-imager depends on qml-module-qtquick-window2; however:
  Package qml-module-qtquick-window2 is not installed.
 rpi-imager depends on qml-module-qtgraphicaleffects; however:
  Package qml-module-qtgraphicaleffects is not installed.

dpkg: error processing package rpi-imager (--install):
 dependency problems - leaving unconfigured
Processing triggers for mime-support (3.64ubuntu1) ...
Errors were encountered while processing:
 rpi-imager
```

maxnet, 2023-01-07:
No idea how well it runs under WSL.

But "dpkg -i some.deb" does not automatically install dependencies.
"apt install ./some.deb" is usually easier.

mhangaard, 2023-02-13:
I managed to get the installation completed by installing the dependencies, but I still do not have a desktop and I think that is needed to use rpi-imager.

closed 2023-02-13 by mhangaard