raspberrypi/rpi-imager #536: fails to build a deb package on bookworm latest qt 5.15.8

xevilstar, 2023-01-14:
Dear mantainer

I was trying to upload a package of the current git to debian mentors

Please add support for qt 5.15.8

```
root@ghost:/usr/src/rpi-imager# apt-get full-upgrade
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
root@ghost:/usr/src/rpi-imager# apt-get full-upgrade
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
```

```
root@ghost:/usr/src/rpi-imager# qmake --version
QMake version 3.1
Using Qt version 5.15.8 in /usr/lib/x86_64-linux-gnu
```

```
root@ghost:/usr/src/rpi-imager# apt-get install libarchive-dev qtdeclarative5-dev libqt5svg5-dev qttools5-dev qml-module-qtquick2 qml-module-qtquick-controls2 qml-module-qtquick-layouts qml-module-qtquick-templates2 qml-module-qtquick-window2 qml-module-qtgraphicaleffects
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:
```

```
The following packages have unmet dependencies:
 libqt5designer5 : Depends: qtbase-abi-5-15-7
 qttools5-dev : Depends: qttools5-dev-tools (= 5.15.7-2)
E: Unable to correct problems, you have held broken packages.
root@ghost:/usr/src/rpi-imager# dpkg-buildpackage -j`nproc` -us -uc
dpkg-buildpackage: info: source package rpi-imager
dpkg-buildpackage: info: source version 1.7.4
dpkg-buildpackage: info: source distribution unstable
dpkg-buildpackage: info: source changed by Floris Bos <bos@je-eigen-domein.nl>
dpkg-buildpackage: info: host architecture amd64
 dpkg-source --before-build .
dpkg-checkbuilddeps: error: Unmet build dependencies: libarchive-dev qtdeclarative5-dev libqt5svg5-dev qttools5-dev qml-module-qtquick2 qml-module-qtquick-controls2 qml-module-qtquick-layouts qml-module-qtquick-templates2 qml-module-qtquick-window2 qml-module-qtgraphicaleffects
dpkg-buildpackage: warning: build dependencies/conflicts unsatisfied; aborting
dpkg-buildpackage: warning: (Use -d flag to override.)
```
```
root@ghost:/usr/src/rpi-imager# cat /etc/debian_version 
bookworm/sid
```

thanks

xevilstar, 2023-01-14:
temporary fix:

```
root@ghost:/home/xevilstar/Downloads# dpkg --ignore-depends=libqt5qml5 --ignore-depends=qml-module-qtquick2 --ignore-depends=qml-module-qtquick-controls2 --ignore-depends=qml-module-qtquick-layouts --ignore-depends=qml-module-qtquick-templates2 --ignore-depends=qml-module-qtquick-window2 --ignore-depends=qml-module-qtgraphicaleffects -i imager_1.7.3_amd64.deb 
(Reading database ... 1269317 files and directories currently installed.)
Preparing to unpack imager_1.7.3_amd64.deb ...
Unpacking rpi-imager (1.7.3) over (1.7.3) ...
Setting up rpi-imager (1.7.3) ...
Processing triggers for gnome-menus (3.36.0-1.1) ...
Processing triggers for bamfdaemon (0.5.6+repack-1) ...
Rebuilding /usr/share/applications/bamf-2.index...
Processing triggers for mate-menus (1.26.0-3) ...
Processing triggers for mailcap (3.70+nmu1) ...
Processing triggers for desktop-file-utils (0.26-1) ...
Processing triggers for hicolor-icon-theme (0.17-2) ...
root@ghost:/home/xevilstar/Downloads# rpi-imager 
rpi-imager: error while loading shared libraries: libQt5Qml.so.5: cannot open shared object file: No such file or directory
root@ghost:/home/xevilstar/Downloads# apt-file search libQt5Qml.so.5
libqt5qml5: /usr/lib/x86_64-linux-gnu/libQt5Qml.so.5
libqt5qml5: /usr/lib/x86_64-linux-gnu/libQt5Qml.so.5.15
libqt5qml5: /usr/lib/x86_64-linux-gnu/libQt5Qml.so.5.15.7
libqt5qml5: /usr/lib/x86_64-linux-gnu/libQt5Qml.so.5.15.8
nsight-compute: /usr/lib/nsight-compute/host/linux-desktop-glibc_2_11_3-x64/libQt5Qml.so.5
nsight-systems: /usr/lib/nsight-systems/host-linux-x64/libQt5Qml.so.5
root@ghost:/home/xevilstar/Downloads# apt-get install libqt5qml5
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
You might want to run 'apt --fix-broken install' to correct these.
The following packages have unmet dependencies:
 rpi-imager : Depends: qml-module-qtquick2 but it is not going to be installed
              Depends: qml-module-qtquick-controls2 but it is not going to be installed
              Depends: qml-module-qtquick-layouts but it is not going to be installed
              Depends: qml-module-qtquick-templates2 but it is not going to be installed
              Depends: qml-module-qtquick-window2 but it is not going to be installed
              Depends: qml-module-qtgraphicaleffects but it is not going to be installed
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).
root@ghost:/home/xevilstar/Downloads# apt --fix-broken install
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Correcting dependencies... Done
The following additional packages will be installed:
  libqt5qml5 libqt5qmlmodels5 libqt5qmlworkerscript5 libqt5quick5 libqt5quickcontrols2-5 libqt5quicktemplates2-5 qml-module-qtgraphicaleffects qml-module-qtquick-controls2 qml-module-qtquick-layouts qml-module-qtquick-templates2 qml-module-qtquick-window2 qml-module-qtquick2
Suggested packages:
  qt5-qmltooling-plugins
The following NEW packages will be installed:
  libqt5qml5 libqt5qmlmodels5 libqt5qmlworkerscript5 libqt5quick5 libqt5quickcontrols2-5 libqt5quicktemplates2-5 qml-module-qtgraphicaleffects qml-module-qtquick-controls2 qml-module-qtquick-layouts qml-module-qtquick-templates2 qml-module-qtquick-window2 qml-module-qtquick2
0 upgraded, 12 newly installed, 0 to remove and 0 not upgraded.
Need to get 4793 kB of archives.
After this operation, 21.8 MB of additional disk space will be used.
Do you want to continue? [Y/n] 
Get:1 http://deb.debian.org/debian sid/main amd64 libqt5qml5 amd64 5.15.8+dfsg-2 [1357 kB]
Get:2 http://deb.debian.org/debian sid/main amd64 libqt5qmlmodels5 amd64 5.15.8+dfsg-2 [190 kB]
Get:3 http://deb.debian.org/debian sid/main amd64 libqt5quick5 amd64 5.15.8+dfsg-2 [1591 kB]
Get:4 http://deb.debian.org/debian sid/main amd64 qml-module-qtquick-window2 amd64 5.15.8+dfsg-2 [27.8 kB]
Get:5 http://deb.debian.org/debian sid/main amd64 libqt5qmlworkerscript5 amd64 5.15.8+dfsg-2 [34.8 kB]
Get:6 http://deb.debian.org/debian sid/main amd64 qml-module-qtquick2 amd64 5.15.8+dfsg-2 [36.4 kB]
Get:7 http://deb.debian.org/debian sid/main amd64 qml-module-qtgraphicaleffects amd64 5.15.8-2 [68.9 kB]
Get:8 http://deb.debian.org/debian sid/main amd64 libqt5quicktemplates2-5 amd64 5.15.8+dfsg-2 [397 kB]
Get:9 http://deb.debian.org/debian sid/main amd64 libqt5quickcontrols2-5 amd64 5.15.8+dfsg-2 [62.7 kB]
Get:10 http://deb.debian.org/debian sid/main amd64 qml-module-qtquick-templates2 amd64 5.15.8+dfsg-2 [70.4 kB]
Get:11 http://deb.debian.org/debian sid/main amd64 qml-module-qtquick-controls2 amd64 5.15.8+dfsg-2 [903 kB]
Get:12 http://deb.debian.org/debian sid/main amd64 qml-module-qtquick-layouts amd64 5.15.8+dfsg-2 [53.9 kB]
Fetched 4793 kB in 1s (3490 kB/s)                      
Selecting previously unselected package libqt5qml5:amd64.
(Reading database ... 1269317 files and directories currently installed.)
Preparing to unpack .../00-libqt5qml5_5.15.8+dfsg-2_amd64.deb ...
Unpacking libqt5qml5:amd64 (5.15.8+dfsg-2) ...
Selecting previously unselected package libqt5qmlmodels5:amd64.
[... 22 lines cut ...]
Unpacking libqt5quickcontrols2-5:amd64 (5.15.8+dfsg-2) ...
Selecting previously unselected package qml-module-qtquick-templates2:amd64.
Preparing to unpack .../09-qml-module-qtquick-templates2_5.15.8+dfsg-2_amd64.deb ...
Unpacking qml-module-qtquick-templates2:amd64 (5.15.8+dfsg-2) ...
[... cut ...]
Setting up qml-module-qtquick-window2:amd64 (5.15.8+dfsg-2) ...
Setting up qml-module-qtquick-layouts:amd64 (5.15.8+dfsg-2) ...
Setting up libqt5quicktemplates2-5:amd64 (5.15.8+dfsg-2) ...
Setting up qml-module-qtquick2:amd64 (5.15.8+dfsg-2) ...
Setting up qml-module-qtquick-templates2:amd64 (5.15.8+dfsg-2) ...
Setting up libqt5quickcontrols2-5:amd64 (5.15.8+dfsg-2) ...
Setting up qml-module-qtgraphicaleffects:amd64 (5.15.8-2) ...
Setting up qml-module-qtquick-controls2:amd64 (5.15.8+dfsg-2) ...
Processing triggers for libc-bin (2.36-8) ...

```

xevilstar, 2023-01-14:
also ... please consider using qt6

maxnet, 2023-01-14:
>fails to build a deb package on bookworm latest qt 5.15.8

Bookworm has Qt 5.15.7
https://packages.debian.org/bookworm/qtbase5-dev

>Package: qtbase5-dev (5.15.7+dfsg-2)

So I am guessing you have sid/unstable in your apt.sources instead.
If common packages like qttools5-dev cannot be installed there, that is a Debian problem, not an Imager problem.

maxnet, 2023-01-14:
>also ... please consider using qt6

Problem is that does not run on certain OSes that quite a few users still use.
Like Windows 8, Ubuntu 20, Mac OS X 10.13

So not good as default for now.
Although Imager do is known to compile under Qt 6 if you patch CMakeLists.txt (Qt5 -> Qt6)

xevilstar, 2023-01-14:
qmake --version
QMake version 3.1
Using Qt version 5.15.8 in /usr/lib/x86_64-linux-gnu
cannot debuild

maxnet, 2023-01-14:
And your /etc/apt/sources.list says bookworm, or does it say something else?

maxnet, 2023-01-14:
>Get:1 http://deb.debian.org/debian sid/main amd64 libqt5qml5 amd64 5.15.8+dfsg-2 [1357 kB]

Isn't "sid" the unstable version AFTER bookworm?
Now that bookworm is "testing"

xevilstar, 2023-01-15:
just a fair warning... from what is happening in the various forums bookworm code freeze is coming. So this is the next stable (I am following the development), I am just a step ahead. This code should be ready for that. It is already strange that a wonderful package I use everyday even at work is not found by apt-cache, that's why I tried to submit it to debianmentors

maxnet, 2023-01-15:
We are not really interested in having this package in Debian, as then people will be stuck with the same old Imager version for 2 years, due to Debian's policy of not allowing version upgrades (other than stability/security backports) within a release.

As far as your problems building it goes, that the Qt 5 packages are broken in Sid is 100÷ a Debian problem.
No Qt project that uses cmake as build system and has translations is going to work without the tools package that we pull in.

closed 2023-01-15 by maxnet

xevilstar, 2023-01-15:
this temporary workaround makes the official package work in debian.

> temporary fix: using current rpi-imager deb downloaded from raspberry site
> 
> ```
> root@ghost:/home/xevilstar/Downloads# dpkg --ignore-depends=libqt5qml5 --ignore-depends=qml-module-qtquick2 --ignore-depends=qml-module-qtquick-controls2 --ignore-depends=qml-module-qtquick-layouts --ignore-depends=qml-module-qtquick-templates2 --ignore-depends=qml-module-qtquick-window2 --ignore-depends=qml-module-qtgraphicaleffects -i imager_1.7.3_amd64.deb 
> (Reading database ... 1269317 files and directories currently installed.)
> Preparing to unpack imager_1.7.3_amd64.deb ...
> Unpacking rpi-imager (1.7.3) over (1.7.3) ...
> Setting up rpi-imager (1.7.3) ...
> Processing triggers for gnome-menus (3.36.0-1.1) ...
> Processing triggers for bamfdaemon (0.5.6+repack-1) ...
> Rebuilding /usr/share/applications/bamf-2.index...
> Processing triggers for mate-menus (1.26.0-3) ...
> Processing triggers for mailcap (3.70+nmu1) ...
> Processing triggers for desktop-file-utils (0.26-1) ...
> Processing triggers for hicolor-icon-theme (0.17-2) ...
> root@ghost:/home/xevilstar/Downloads# rpi-imager 
> rpi-imager: error while loading shared libraries: libQt5Qml.so.5: cannot open shared object file: No such file or directory
> root@ghost:/home/xevilstar/Downloads# apt-file search libQt5Qml.so.5
> libqt5qml5: /usr/lib/x86_64-linux-gnu/libQt5Qml.so.5
> libqt5qml5: /usr/lib/x86_64-linux-gnu/libQt5Qml.so.5.15
> libqt5qml5: /usr/lib/x86_64-linux-gnu/libQt5Qml.so.5.15.7
> libqt5qml5: /usr/lib/x86_64-linux-gnu/libQt5Qml.so.5.15.8
> nsight-compute: /usr/lib/nsight-compute/host/linux-desktop-glibc_2_11_3-x64/libQt5Qml.so.5
> nsight-systems: /usr/lib/nsight-systems/host-linux-x64/libQt5Qml.so.5
> root@ghost:/home/xevilstar/Downloads# apt-get install libqt5qml5
> Reading package lists... Done
> Building dependency tree... Done
> Reading state information... Done
> You might want to run 'apt --fix-broken install' to correct these.
> The following packages have unmet dependencies:
>  rpi-imager : Depends: qml-module-qtquick2 but it is not going to be installed
>               Depends: qml-module-qtquick-controls2 but it is not going to be installed
>               Depends: qml-module-qtquick-layouts but it is not going to be installed
>               Depends: qml-module-qtquick-templates2 but it is not going to be installed
>               Depends: qml-module-qtquick-window2 but it is not going to be installed
>               Depends: qml-module-qtgraphicaleffects but it is not going to be installed
> E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).
> root@ghost:/home/xevilstar/Downloads# apt --fix-broken install
> Reading package lists... Done
> Building dependency tree... Done
> Reading state information... Done
> Correcting dependencies... Done
> The following additional packages will be installed:
>   libqt5qml5 libqt5qmlmodels5 libqt5qmlworkerscript5 libqt5quick5 libqt5quickcontrols2-5 libqt5quicktemplates2-5 qml-module-qtgraphicaleffects qml-module-qtquick-controls2 qml-module-qtquick-layouts qml-module-qtquick-templates2 qml-module-qtquick-window2 qml-module-qtquick2
> Suggested packages:
>   qt5-qmltooling-plugins
> The following NEW packages will be installed:
>   libqt5qml5 libqt5qmlmodels5 libqt5qmlworkerscript5 libqt5quick5 libqt5quickcontrols2-5 libqt5quicktemplates2-5 qml-module-qtgraphicaleffects qml-module-qtquick-controls2 qml-module-qtquick-layouts qml-module-qtquick-templates2 qml-module-qtquick-window2 qml-module-qtquick2
> 0 upgraded, 12 newly installed, 0 to remove and 0 not upgraded.
> Need to get 4793 kB of archives.
> After this operation, 21.8 MB of additional disk space will be used.
> Do you want to continue? [Y/n] 
> Get:1 http://deb.debian.org/debian sid/main amd64 libqt5qml5 amd64 5.15.8+dfsg-2 [1357 kB]
> Get:2 http://deb.debian.org/debian sid/main amd64 libqt5qmlmodels5 amd64 5.15.8+dfsg-2 [190 kB]
> Get:3 http://deb.debian.org/debian sid/main amd64 libqt5quick5 amd64 5.15.8+dfsg-2 [1591 kB]
> Get:4 http://deb.debian.org/debian sid/main amd64 qml-module-qtquick-window2 amd64 5.15.8+dfsg-2 [27.8 kB]
> Get:5 http://deb.debian.org/debian sid/main amd64 libqt5qmlworkerscript5 amd64 5.15.8+dfsg-2 [34.8 kB]
> Get:6 http://deb.debian.org/debian sid/main amd64 qml-module-qtquick2 amd64 5.15.8+dfsg-2 [36.4 kB]
> Get:7 http://deb.debian.org/debian sid/main amd64 qml-module-qtgraphicaleffects amd64 5.15.8-2 [68.9 kB]
> Get:8 http://deb.debian.org/debian sid/main amd64 libqt5quicktemplates2-5 amd64 5.15.8+dfsg-2 [397 kB]
> Get:9 http://deb.debian.org/debian sid/main amd64 libqt5quickcontrols2-5 amd64 5.15.8+dfsg-2 [62.7 kB]
> Get:10 http://deb.debian.org/debian sid/main amd64 qml-module-qtquick-templates2 amd64 5.15.8+dfsg-2 [70.4 kB]
> Get:11 http://deb.debian.org/debian sid/main amd64 qml-module-qtquick-controls2 amd64 5.15.8+dfsg-2 [903 kB]
> Get:12 http://deb.debian.org/debian sid/main amd64 qml-module-qtquick-layouts amd64 5.15.8+dfsg-2 [53.9 kB]
> Fetched 4793 kB in 1s (3490 kB/s)                      
> Selecting previously unselected package libqt5qml5:amd64.
> (Reading database ... 1269317 files and directories currently installed.)
> Preparing to unpack .../00-libqt5qml5_5.15.8+dfsg-2_amd64.deb ...
[... 23 lines cut ...]
> Preparing to unpack .../08-libqt5quickcontrols2-5_5.15.8+dfsg-2_amd64.deb ...
> Unpacking libqt5quickcontrols2-5:amd64 (5.15.8+dfsg-2) ...
[... cut ...]
> Setting up qml-module-qtquick-layouts:amd64 (5.15.8+dfsg-2) ...
> Setting up libqt5quicktemplates2-5:amd64 (5.15.8+dfsg-2) ...
> Setting up qml-module-qtquick2:amd64 (5.15.8+dfsg-2) ...
> Setting up qml-module-qtquick-templates2:amd64 (5.15.8+dfsg-2) ...
> Setting up libqt5quickcontrols2-5:amd64 (5.15.8+dfsg-2) ...
> Setting up qml-module-qtgraphicaleffects:amd64 (5.15.8-2) ...
> Setting up qml-module-qtquick-controls2:amd64 (5.15.8+dfsg-2) ...
> Processing triggers for libc-bin (2.36-8) ...
> ```

maxnet, 2023-01-15:
>this temporary workaround makes the official package work in debian.

If you want to install our .deb package, downloading it and executing "sudo apt install ./*.deb" is very likely to just work.
You are maling the mistake of using dpkg which does not install dependencies automatically...

That is a different issue than you have when trying to build Imager, which fails because the qt tools package is broken on Sid.
That package is only needed during building Imager and is used by cmake to build the translation files to have Imager in multiple langauges.
It is not needed after build, so that packags being broken om Sid does not prevent you using our prebuild .deb.

xevilstar, 2023-01-15:
usually I use gdebi but it failed on debian because of old deps. the workaround fixed the issue

maxnet, 2023-01-15:
>usually I use gdebi but it failed on debian because of old deps. the workaround fixed the issue

Then gdebi probably failed to do the equivalent of "apt update" first, to update its local copy of the repository information first...
You are using an unstable development version of Debian (sid), and those files change a lot on that...

Your "workaround" of first using dpkg with "--ignore-depends=name-of-package" and then calling "apt --fix-broken install" to install those exact dependencies you have been ignoring first does not make much sense.
If "apt --fix-broken" can find the dependencies after all, so could "sudo apt install ./imager*.deb".