raspberrypi/rpi-imager #537: [OS request] [URL update] Lakka

ToKe79, 2023-01-15:
Dear mainteners,

I kindly ask to update the URLs in the snippet for the Games and Emulation OS "Lakka" due to changes in the infrastructure.

TL;DR; Substitute `nightly` with `le` in the `icon` and `subitems_url` keys

Current snippet:
```
 "os_list": [
        {
            "name": "Lakka",
            "description": "The DIY retro emulation console",
            "icon": "https://nightly.builds.lakka.tv/.rpi-imager/icon.png",
            "website": "https://lakka.tv/",
            "subitems_url": "https://nightly.builds.lakka.tv/.rpi-imager/rpi-imager.json"
        }
    ]
```
Updated snippet:
```
 "os_list": [
        {
            "name": "Lakka",
            "description": "The DIY retro emulation console",
            "icon": "https://le.builds.lakka.tv/.rpi-imager/icon.png",
            "website": "https://lakka.tv/",
            "subitems_url": "https://le.builds.lakka.tv/.rpi-imager/rpi-imager.json"
        }
    ]
```

Thank you!

ToKe79, 2024-03-28:
Hi, I just noticed that Lakka does not show up in the Emulation and game OS section of the Raspberry Pi Imager. Was there any reason for removing the distro?

tdewey-rpi, 2024-09-19:
> Hi, I just noticed that Lakka does not show up in the Emulation and game OS section of the Raspberry Pi Imager. Was there any reason for removing the distro?

The most likely reason is that Lakka failed to respond to the request for a contact point that I filed last year.

A representative from Lakka would need to request a new OS addition using the form: https://forms.gle/sG6V6trq3shqr26UA

Closing this issue - this isn't an issue with `rpi-imager`, but with the OS list.

closed 2024-09-19 by tdewey-rpi