raspberrypi/rpi-imager #538: Feature Request: Estimated remaining time

zigpot, 2023-01-16:
I use rpi imager to write images to a storage. This process is often long, so an estimation of the remaining time during the process would be very much useful.

tdewey-rpi, 2024-02-07:
Feature request rejected.

As anyone who used Windows in the 90s may recall, "Estimated time remaining" is an incredibly difficult thing to actually calculate with any acceptable degree of precision or accuracy. In reality, it's a factor of all _sorts_ of variables - current DVFS state of the CPU, RAM and bus contention, the precise connection topology, what else is going on in the system, etc. To produce something I'd be happy with shipping would probably change the main focus of this project by time away from helping users set up their Raspberry Pi devices to becoming an effective bandwidth estimator for a range of USB devices.

Unfortunately I'm not prepared to commit to that level of change for a feature that, while cool, might only be visible for a relatively short time.

closed 2024-02-07 by tdewey-rpi

lurch, 2024-02-07:
...also: network download bandwidth, load on the server that you're downloading the image from, ...