raspberrypi/rpi-imager #539: Proxy (with authentication support) on Windows

MisterAnderson91, 2023-01-16:
Currently the rpi-imager app does not seem to detect or respect an http proxy requiring authentication on at least Windows 10.

Unfortunately these proxy systems are still common in education environments such as schools.

Although the workaround of downloading zip files of images is ok, it would be appreciated if changes could be made to allow direct downloads.

The current behaviour of rpi-imager in these environments is the same as when there is no internet access, "Error downloading OS list from internet". The URL https://downloads.raspberrypi.org/os_list_imagingutility_v3.json is accessible in a browser.

maxnet, 2023-01-25:
>Currently the rpi-imager app does not seem to detect or respect an http proxy requiring authentication on at least 
>Windows 10.

There do is code in the Qt framework used that is supposed to grab the proxy settings from the operating system.
But I am not sure how well that works in environments that have it configured that the Windows username and password of the logged in user are used automatically.
That Imager runs at elevated privileges as administrative user instead of the current user, could interfere with that as well. Recall there is also a somewhat related issue that network shares the current user normally has access to, may not work properly under that environment.
Not sure if there is a good fix for this, that does not require entering authentication details manually...

MisterAnderson91, 2023-01-26:
This proxy doesn't source the username and password from the current user, it always requires manually entered credentials. However it also never asks for the credentials manually either, which is the issue.