raspberrypi/rpi-imager #541: Ampersand in Wifi-password

AndreasHoffmann2, 2023-01-22:
Hi,

my Wifi-password contains an ampersand. When prefilling the password from my Windows-computer's settings, it does encode the ampersand:
[image]

After writing this to the sd-card, the raspi is unable to connect to the Wifi.

The correct password would have been:
asdklfjweiormd&ads

This is not my real Wifi-password of course, just an example for the actual problem.

maxnet, 2023-01-22:
Duplicate of #486
Fixed in 1.7.4

maxnet, 2023-01-22:
My bad, ampersand not apostrophe.
Reopening.

AndreasHoffmann2, 2023-01-22:
Thank you for reopening the issue.
Just to avoid further misunderstandings about the issue, please find the steps to reproduce it below:
1.) Connect your Windows-machine to a Wifi containing an ampersand in the password (let's assume 123&456).
2.) Start "Raspberry Pi Imager v1.7.3"
3.) Open the settings. The SSID and password for the Wifi should be prefilled.
4.) Check "show password"

Actual behaviour:
The password is shown as 123&amp;456. After writing this to the sd-card, the Raspi will not be able to connect to Wifi.

Expected behaviour:
The password should be correct.

maxnet, 2023-01-25:
Can you confirm the following version fixes the issue for your network?

[imager-1.7.4.exe](https://www.maxnet.eu/tmp/174-fat/imager-1.7.4.exe)

AndreasHoffmann2, 2023-01-26:
Thank you, that fixed it.

closed 2023-03-24 by maxnet