raspberrypi/rpi-imager #542: Operating system did not mount fat32 partition

aikonen, 2023-01-24:
Using windows 10 and attempting to write using rpi imager, I am getting a FAT32 partitioning error. After manually setting the partition using the minitool partition wizard, I still come across the same error code.

andreabusi, 2023-01-31:
Same error here, macOS 13.2.

I was able to write images with macOS 13.1, now it seems broken.

maxnet, 2023-01-31:
>Same error here, macOS 13.2.

Can you try if this one works better?
[Raspberry Pi Imager 1.7.4.dmg](https://www.maxnet.eu/tmp/174-xml/Raspberry%20Pi%20Imager%201.7.4.dmg)

andreabusi, 2023-01-31:
> > Same error here, macOS 13.2.
> 
> Can you try if this one works better? [Raspberry Pi Imager 1.7.4.dmg](https://www.maxnet.eu/tmp/174-xml/Raspberry%20Pi%20Imager%201.7.4.dmg)

It works!

NovaViper, 2023-02-06:
Is there a version of this for Linux? Having the same issues trying to flash an usb ssd for my pi 4

maxnet, 2023-02-06:
> Is there a version of this for Linux? Having the same issues trying to flash an usb ssd for my pi 4

Only for Intel/amd64

https://www.maxnet.eu/tmp/174-xml/rpi-imager_1.7.4_amd64.deb

closed 2023-03-24 by maxnet

FlorianCassayre, 2023-07-22:
I confirm that updating `rpi-imager` solves the problem. I had version `1.7.2` and I would consistently encounter this problem when using advanced settings. Upgrading to `1.7.5` solved the problem.

Note that I had to install that version from the `.deb` package as `apt` did not have the latest version available in the registry.

mariuszbrz, 2023-09-09:
I had the same problem on linux machine. Below link to my comment how to fix it:
https://github.com/raspberrypi/rpi-imager/issues/505#issuecomment-1712471230

SengerM, 2024-04-03:
Same problem here (Ubuntu 22.04), solved [as pointed out by FlorianCassayre](https://github.com/raspberrypi/rpi-imager/issues/542#issuecomment-1646598337). For quick reference:

1. Uninstall your `rpi-imager` with `sudo apt remove rpi-imager`.
2. Manually download the latest version from [the official downloads link](https://www.raspberrypi.com/software/).
3. Manually install it with `sudo apt install ./imager_1.8.5_amd64.deb`.
4. Enjoy.