raspberrypi/rpi-imager #543: [OS request] moOde audio player image

moodeaudio, 2023-01-25:
Hi,

Our project would like to have moOde audio player included in the Media player OS section. The entry for the top-level json file is below.

```
{
	"name": "moOde audio player",
	"description": "Audiophile music streamer for the Raspberry Pi",
	"website": "https://moodeaudio.org",
	"icon": "https://moodeaudio.org/assets/img/logo/moode-logo-d.png",
	"subitems_url": "https://moodeaudio.org/rpi-imager-subitems.json"
}
```
The links in the top and subitems json have been tested locally using --repo.

[image]

-Tim

XECDesign, 2023-01-30:
I think the main problem here is that there's ssh on by default, with root access and default passwords. Seems like before SSH can be turned on, there should be some kind of setup that forces a password change for the user and for the WLAN AP.

moodeaudio, 2023-01-30:
SSH is on but root user is disabled. Only the user pi exists by default in our build but yes having remote access on and default passwords is not the best.

We have a TODO item to address that in our next release which is scheduled for Q1 2023. The planned implementation is similar to what you are proposing.

What exactly is the approval process?

XECDesign, 2023-01-30:
> SSH is on but root user is disabled. Only the user pi exists by default in our build but yes having remote access on and default passwords is not the best.

I was able to get root access by getting in through SSH and running `sudo -i`.

> What exactly is the approval process?

There isn't one, everything is on a case by case basis. If the distro looks useful (it does), there are no major issues (just what has been mentioned already) and developers don't get people to brigade the issue to make a distro look bigger than it is, then it will go in.

moodeaudio, 2023-01-30:
Ok sounds good. I'll update the issue when we publish our next release and we'll go from there.

moodeaudio, 2023-03-15:
Hi,

Our 8.3.0 release has just shipped and the image has no SSH or any userid's or passwords. Our users are directed to use the official Pi Imager to set these up.

Here's a link to the release announcement.
https://moodeaudio.org/forum/showthread.php?tid=5430&pid=45686#pid45686

-Tim

XECDesign, 2023-03-16:
Many thanks! I'll take a look next week.

XECDesign, 2023-03-21:
Added

moodeaudio, 2023-03-21:
Great, thanks :-)

moodeaudio, 2023-03-26:
Hi,

Some of our user have reported a few issues:

1. moOde is listed in the "Other specific-purpose OS" section.
It should be listed in the "Media Player OS" section

2. When selecting the OS from "Other specific-purpose OS" section the COG icon does not appear and thus no Advanced options can be entered.

Here are the item and subitem JSON's at our website in case we need to make some modifications.

-Tim

rpi-imager-item.json
```
{
	"name": "moOde audio player",
	"description": "Audiophile music streamer for the Raspberry Pi",
	"website": "https://moodeaudio.org",
	"icon": "https://moodeaudio.org/assets/img/logo/moode-logo-d.png",
	"subitems_url": "https://moodeaudio.org/rpi-imager-subitems.json"
}
```
rpi-imager-subitems.json
```
{
"os_list": [
		{
			"url": "https://github.com/moode-player/moode/releases/download/r830prod/image_2023-03-14-moode-r830-arm64-lite.zip",
			"extract_size": 3649044480,
			"extract_sha256": "64e12e70bbbd647fdbcebc38d1e57866e4fce947104bd71d55c560f4e0f718a4",
			"image_download_size": 1075293774,
			"description": "Audiophile music streamer for the Raspberry Pi",
			"icon": "https://moodeaudio.org/assets/img/logo/moode-logo-d.png",
			"name": "moOde audio player (64-bit)",
			"release_date": "2023-03-14"
		},
		{
			"url": "https://github.com/moode-player/moode/releases/download/r830prod/image_2023-03-14-moode-r830-lite.zip",
			"extract_size": 3716153344,
			"extract_sha256": "4469e3b19b946fbf0fa28f7778154bc3c89bdd88ad2dcda966820035abfcae03",
			"image_download_size": 1210169305,
			"description": "Audiophile music streamer for the Raspberry Pi",
			"icon": "https://moodeaudio.org/assets/img/logo/moode-logo-d.png",
			"name": "moOde audio player (legacy 32-bit)",
			"release_date": "2023-03-14"
		}
	]
}
```

XECDesign, 2023-03-26:
Moved to the right category.

moodeaudio, 2023-03-26:
Confirmed.

closed 2023-03-26 by moodeaudio