raspberrypi/rpi-imager #545: How to get an OS listed in the RPi Imaging program

Poporacer, 2023-01-28:
I could not find any documentation on how to include an OS into the RPi Imager program?
What is required and what is the process?

XECDesign, 2023-01-30:
If you're the developer of the distro, you'd need to open an issue.

https://github.com/raspberrypi/rpi-imager/issues?q=label%3A"OS+request"

Include the json file snippet you would like included in https://downloads.raspberrypi.org/os_list_imagingutility_v3.json

The `subitems_url` field needs to point to an official server.

The distro needs to differentiate itself from what's already offered. For example, it can't be "Raspberry Pi OS, but I've installed some stuff I like".

macmpi, 2023-02-20:
Is there some sort of baseline how-to doc for repo maintainers to describe the various requirements & settings options to craft necessary image and .json files?
- is image format mandatory, which form (i.e, is a simple directory archive also ok with rpi-imager creating FAT partition on its own)?
- which .json keywords are mandatory/optional
- what does init-format option does; which are acceptable values
- ...

Maybe a recommendation (enforcement?) [to not label boot partition as `boot`](https://github.com/RPi-Distro/pi-gen/issues/669) could take place in-there too :wink:

Thanks for consideration.

tdewey-rpi, 2023-11-17:
Please fill out the form to request an addition to the OS list at https://forms.gle/cbsU27aLLVaxUXbw8.

closed 2023-11-17 by tdewey-rpi