raspberrypi/rpi-imager #546: No third-party images available for writing

ok11, 2023-01-28:
As said, no Home assistant, no Octoprint, no Homebridge, nothing, everything ends up with "Error downloading oS list from internet". This way I doubt that some of the provider hosts are down. 
No blacklisting or child protection controls are active on the network.
Could you please check what is going on?

maxnet, 2023-01-28:
Which OS are you running Imager on?
Your computer does have correct date/time/timezone settings and you do are running an up-to-date operating system version that is not end-of-life?

As "Error downloading" can also mean a failure to verify SSL certificates.
Which can happen for various reasons, for example if your operating system does not have the certificate of the certificate authority used by the remote webserver installed.
If you are using a web browser that is using its own list (and not the one of the operating system), that indeed do can work.

ok11, 2023-01-28:
First of all, thanks for the quick reply.
I'm on windows 10, and its EOL is far from having being reached ;)

I also have doubts regarding TLS certs, unless the endpoint using something exotic as a CA. All standard CAs (DigiCert, VeriSign, even Let's Encrypt are there, as provided as a standard browser content; it is Chrome  109.0.5414.75 BTW)...

maxnet, 2023-01-28:
If you go the Window start menu, enter "cmd", and then RIGHT CLICK the command prompt icon and select "Run as admin" and enter: 

```
cd \program files (x86)\Raspberry Pi Imager
set QML_XHR_DUMP=1
rpi-imager --debug
```

Does it output any HTTP errors to console?

(Running the command prompt as admin user is important, otherwise the "set QML_XHR_DUMP=1" will not be effective, as elevated programs do not take environment variables from normal users)

ok11, 2023-01-28:
Tried that just to find out that it works now :)
I will do the debugging when (and if) it repeats, for now closing the issue.

Thanks a lot for supporting!

closed 2023-01-28 by ok11