raspberrypi/rpi-imager #548: [OS Request] Falcon Player (FPP)

darylc, 2023-02-01:
Hi,

I am a developer for Falcon Player (FPP).  We would like to be added to rpi-imager, under the 'Other specific-purpose OS' category please.

                {
                    "name": "Falcon Player (FPP)",
                    "description": "Falcon Player is a feature-rich sequence player/controller interface designed to run on low-cost Single Board Computers for animated lighting shows",
                    "icon": "https://raw.githubusercontent.com/FalconChristmas/fpp/master/rpi-imager/rpi-imager_fpp_logo.png",
                    "website": "https://falconchristmas.github.io/",
                    "subitems_url": "https://raw.githubusercontent.com/FalconChristmas/fpp/master/rpi-imager/rpi-imager_falcon_player.json"
                }

Thanks for your consideration.

ghollingworth, 2023-02-01:
It took me quite a long time to work out what this actually did...

Do you have a website that explains what it is and why someone might be interested?  The idea of the weblink is to provide people with a little more information but yours only really has "sequence player" in the description.

Most users of Raspberry Pi will not know what a sequence player is where it would be used or where it might otherwise be useful.

"

Poporacer, 2023-02-01:
We don't have a specific website for this application, since in this hobby the application is well known, but it is used world wide for animated lights shows like you see on The Great Christmas Light Fight, but it is used for many other things as well such as landscape lighting, whole house lighting, carnivals, etc. We will update the  Web link to provide more information. We allow the user to share anonymous statistics for troubleshooting/development and we have no way of telling how many people opt in , but based on those that have, we have over 16,000 devices with this application.
This application has a lot of functionality. 
It is used to automatically start/stop your light show based on a schedule to include starting/stopping your show relative to your local sunrise sunset, and use multiple different playlists for different occasions. It also has a controller functionality. We will update the webpage ling to provide more information. 
Here is an example of a lightshow using FPP https://harrischristmas.com/2020Videos/Mary_Did_You_Know.mp4

ghollingworth, 2023-02-01:
Yeah, throw all those things into your github.io site with the idea that someone visiting doesn't know what it is at all (at least to cover the first couple of paragraphs).  

For Raspberry Pi, it's great to have people doing interesting and valuable things with it in the open source scene because it showcases our capability and shows just how wide the support is, that's actually why we've enabled people to add their images to Imager.

Gordon

Poporacer, 2023-02-01:
The github.io site has been updated and is being reviewed for additional information. Is this what you were looking for before we go down the wrong path?

ghollingworth, 2023-02-01:
Yeah, that's great, gives a good introduction to what it's actually for!

Gordon

darylc, 2023-02-07:
Slightly updated json extract:

            {
                "name": "Falcon Player (FPP)",
                "description": "Falcon Player (FPP) is an appliance used for managing and controlling animated light shows",
                "icon": "https://raw.githubusercontent.com/FalconChristmas/fpp/master/rpi-imager/rpi-imager_fpp_logo.png",
                "website": "https://www.falconplayer.com/",
                "subitems_url": "https://raw.githubusercontent.com/FalconChristmas/fpp/master/rpi-imager/rpi-imager_falcon_player.json"
            }

lurch, 2023-02-07:
> `"subitems_url": "https://raw.githubusercontent.com/FalconChristmas/fpp/master/rpi-imager/rpi-imager_falcon_player.json"`

Just out of curiosity, is there a good reason for that listing each of "FPP v6.1.1", "FPP v6.2" & "FPP v6.3", rather than just the latest version? :man_shrugging:

darylc, 2023-02-07:
@lurch Our end-users tend to use multiple instances of Falcon Player and once they are happy with a version they stick with it for a while so we listed our last 3 to make it easier for them to get the exact version they wanted.  If you prefer we can just list the latest or move the older ones into a sub folder?

lurch, 2023-02-07:
Nah, no problem at all; if there's a good reason for also listing older versions, then feel free to leave it as it is :slightly_smiling_face:

cybercop23, 2023-03-20:
Hi @lurch. Do you have any idea when this may be added to the list? We have folks starting to set-up/upgrade their shows and there have been some new show lighting controllers running on PIs, and this will make the setup so much easier.

ghollingworth, 2023-03-21:
@XECDesign Can you add this...

XECDesign, 2023-03-21:
Added

cybercop23, 2023-03-21:
Thank you very much mates. Cheers.