raspberrypi/rpi-imager #550: Invalid character in ssh key makes boot stop

vpaprots, 2023-02-03:
Disclaimer: This is perhaps user error, and if so, feel free to close. But I spent 3 days debugging this, trying sd cards, power supplies and thinking I fried my pi (Since the monitor is off, no feedback on failure). Out of ideas, I put `touch /boot/hello.txt` at the entry of `firstboot.sh`, which is how I realized thats where boot stalled

I had a quote at the end of my ssh-key (copy-paste mistake). `firstrun.sh` escaped the quote with `\'` but the entire string is surrounded with single quotes (that trick only works on double-quotes. As I recall, _nothing_ is escaped in single-quoted string)
```
'ssh-rsa abcd comment\''
```

What I would expect:
- multiline ssh-key field in the GUI to see the entire key
- ssh-key rejected and/or proper quoting/escape
- `firstboot.sh` does `set -x` and outputs to visible tty

maxnet, 2023-02-03:
The escaping of single quotes is fixed in 1.7.4
Related to #541

vpaprots, 2023-02-03:
Thanks @maxnet