raspberrypi/rpi-imager #551: Imager breaks if Comodo revocation checking endpoints are blocked or offline

matt40k, 2023-02-04:
Logging for Scott Helme

He's seeing:
[image]

> _It seems that it's reaching out to the Comodo revocation checking endpoints. This is kind of interesting because if you think about it, when Comodo has down time... nobody can flash a Raspberry Pi microSD card, it's literally that ridiculous!_

https://twitter.com/Scott_Helme/status/1621258437035278342?s=20&t=x_4YcCxFVvVclMOnmrDuGQ

https://scotthelme.co.uk/revocation-checking-is-pointless/

maxnet, 2023-02-04:
Eh, in his blog post he complains that most browsers skip checks if he intently blocks access to the Certificate Revocation List server, and that being insecure, as that is exactly what an attacker with control over network communication would do...

Now he finally found one program that does refuse service when CRL checks cannot be completed, and is now unhappy about that instead???

When was the last time Comodo's CRL server was really down?

matt40k, 2023-02-04:
IDK, just logging tbh cause he didn't after I poked saying hey, maybe log an issue so one of the devs can have a look vs poking the Raspberry Pi social media team - https://twitter.com/Scott_Helme/status/1621258922119012353?s=20&t=kxjqeLPsBMkJDKkve6Yamw

Looks like others have hit issues in locked down environments:
https://github.com/raspberrypi/rpi-imager/issues/433

Guess maybe a config way of reducing the extra security?

maxnet, 2023-02-04:
BTW there do are ways to make things work even when the user has blocked access to the revocation server.
Instead of the client having to contact the certificate authority it may also be possible for the download webserver to do so, and sent a copy of a recent OCSP result to the client as proof certificate is still valid.

Known as OCSP stabling: https://en.m.wikipedia.org/wiki/OCSP_stapling

Do would require not only RPI Ltds's webserver configuration is changed to support this, but also the download servers used by the other Linux distributions we offer for download...
Currently every distribution is providing its own download servers.