raspberrypi/rpi-imager #552: SD card is not showing up in Storage > Choose Storage

voicesofhim, 2023-02-05:
Imager v1.7.3 not displaying SD Card in Storage > Choose Storage. 

Issue is the same when having Raspberry Pi 0S selected or any other OS selected - storage options don't show 

Raspberry Pi Running Fine from SD Card, no issues adding/loading/saving new programs to SD 

Running v1.7.3 on a M1 Mac Pro

maxnet, 2023-02-05:
May need to wiggle the SD card a little bit before the OS (and consequently Imager) detects the card.
The SD card readers in laptops leave to be desired. Better score a USB(-C) reader.

voicesofhim, 2023-02-06:
Thanks. The SD slot is in the Pi 400 itself. The SD is able to store/retrieve data fine when adding new apps, storing pics etc.  The storage doesn't show up as available in the Imager itself for some reason.

maxnet, 2023-02-06:
First you wrote:

>Running v1.7.3 on a M1 Mac Pro

And now you write:

>The SD slot is in the Pi 400 itself.

So did you insert the SD card into the SD card slot that the more expensive Mac Pro (14"/16" models) have, and started Imager on your Mac?

Or are you running Imager on the Pi 400?
Using Imager **under RPI OS** on the Pi 400 will only work if you use an external USB SD card reader. It will refuse to self-destruct the SD card that is currently being used by the operating system.
Although if you have a **wired** Ethernet connection, you do can use the netinstall method instead: https://www.raspberrypi.com/documentation/computers/getting-started.html#installing-over-the-network

voicesofhim, 2023-02-06:
My Fault! Yes the Imager is running on the Pi 400, Mac Pro has nothing to do with it ;) The external USB card reader info makes sense. Appreciate your help!

closed 2023-03-14 by ghollingworth