raspberrypi/rpi-imager #553: [OS Request] ParrotOS

danterolle, 2023-02-07:
Hello everyone! I'm one of the developers of ParrotOS (parrotsec.org) and we have been providing OS [images for Raspberry Pi platforms](https://parrotsec.org/download/?version=raspberry) for a long time.

Since we find it useful the Raspberry Pi Imager, which we also recommend using in our [documentation](https://parrotsec.org/docs/installation/raspberrypi), we would like to ask you to integrate it into your software. 

I understand that you need a file very similar to this: http://downloads.raspberrypi.org/os_list_imaginguutility_v3.json

What is the procedure to generate it? Is there possibly a newer version so I can edit it manually?

Thanks to anyone who will reply.

lurch, 2023-02-07:
See #545 and e.g. #548

danterolle, 2023-02-07:
@lurch Thank you for your answer. I'm waiting for the 5.2 release of our OS and in a few days I'll send everything here.

danterolle, 2023-03-10:
Hello @lurch, I followed the examples as written in those threads. I created a specific [repository](https://gitlab.com/parrotsec/project/parrot-rpi-imager) for it. 

```
{
    "name": "Parrot Security",
    "description": "Parrot Security (ParrotOS, Parrot) is a Free and Open source GNU/Linux distribution based on Debian Stable designed for security experts, developers and privacy aware people.",
    "website": "https://parrotsec.org/",
    "icon": "https://gitlab.com/parrotsec/project/graphics/-/blob/d19ac8a11292136cda4b1df80a2bd156f768fed8/logo/parrot-logo.png",
    "subitems_url": "https://gitlab.com/parrotsec/project/parrot-rpi-imager/-/blob/main/rpi-imager.json"
}
```

[Here](https://gitlab.com/parrotsec/project/parrot-rpi-imager/-/blob/main/rpi-imager.json) is the list of Raspberry Pi images we provide to users. 

Do I need to provide anything else? When will the request be approved? As far as updates are concerned (for example, there might be some new releases in the coming months), who should I notify?

Thank you.

drushtx, 2023-03-14:
@danterolle, did you ever complete or succeed in this effort? We're going to try the same thing to include Rocky Linux images into the imager.

Thank you.

danterolle, 2023-03-14:
Hi @drushtx! As you can see, I'm currently waiting for a response from them. In any case, I think I have correctly provided the information they requested, so for the moment we have to wait.

drushtx, 2023-03-14:
Thank you. I saw that there was no reply but I had hoped that you had "finalized" behind the scenes. We'll keep at it.

danterolle, 2023-04-13:
Hello @lurch @ghollingworth! Is there any news regarding the addition of Parrot to the Raspberry Pi Imager OS list? Thank you

XECDesign, 2023-04-13:
The json file and snippet you provided needs a bit more work.

Currently, the URLs point to gitlab pages rather than the 'raw' resource'. Also, the linked json file has a trailing comma, which is not allowed by the spec, so it fails to parse.

You can download the main json file from [here](https://downloads.raspberrypi.com/os_list_imagingutility_v3.json), edit it to add your snippet and then run `rpi-imager` like this:

```
rpi-imager --repo os_list_imagingutility_v3.json
```

danterolle, 2023-04-17:
Now it should be fixed, what do you think @XECDesign?

```
{
    "name": "Parrot Security",
    "description": "Parrot Security (ParrotOS, Parrot) is a Free and Open source GNU/Linux distribution based on Debian Stable designed for security experts, developers and privacy aware people.",
    "website": "https://parrotsec.org/",
    "icon": "https://deb.parrot.sh/parrot/misc/assets/parrot-logo.png",
    "subitems_url": "https://deb.parrot.sh/parrot/misc/assets/parrot-rpi-imager/rpi-imager.json"
}
```

Thanks for explaining me how to proceed.

danterolle, 2023-11-20:
Hello @tdewey-rpi! Thank you for sending me the link. We are working on the full release of ParrotOS 6 (and consequently also on updating the rpi versions). I will fill out the form as soon as it is ready.