raspberrypi/rpi-imager #558: Advanced dropdowns behaviors

Deadweasel, 2023-02-26:
This is one of those "quality of life" improvement requests.

The country default should be whatever the local system is set to.  The timezone dropdown already does that; why can't this one too?
The reason that's a request is that I live in the US, the default is GB, and the dropdown list only responds to mouse wheel or arrow keys.  I cannot just grab the slider and zoom allllllllll the way down to the U's, and neither can I skip them more quickly with PG UP/DN.  If we can't get the country code to default to the host system settings, could we at least get means of getting around the miles-long list more effectively than "hold down arrow key and wait a little bit"?

Yes yes it's a minor gripe, but also one of those expectations developed from years of common behaviors built into OS and web controls that makes this feel clunky by comparison.

tdewey-rpi, 2024-02-13:
Coming back around to this one - the QtQuick controls we're using enforce this sort of behaviour - unfortunately, they're optimised for Touch rather than keyboard/mouse.

A workaround that you might not have tried is to treat your mouse like it's a hand - "grab" the list and fling it upwards, etc. You'll find it has inertial scrolling.

I assume by "Country dropdown" you mean the Wireless LAN region selector.

Unfortunately auto-detection of this is pretty fraught and not as reliable as you might think. I recently, for example, had to explain to a family member living in England that having their locale set to English/Indian was not going to help with the strange location behaviour they were seeing in other applications. In such a case, we'd potentially wind up setting the WiFi restrictions for an entirely wrong location.

Maintaining the explicit country selector, while cumbersome, does force the user to consider their deployment location.

Regarding the GB default, this is fairly standard across Raspberry Pi maintained code and documentation - we explicitly style for English/British, and I can't see that changing any time soon.

All of that said, I do accept that the current controls are cumbersome to use, and that we could do a better job with selecting location. This is something I've actively looking at resolving.

lurch, 2024-02-13:
Related issues: #519 #352