raspberrypi/rpi-imager #559: Pi imager save the iso?

mirko8054, 2023-03-03:
Hello, as i ask in the title, is there a way to save the iso rather than directly write in a SD Card? This is because i run Pimox in my raspberry and i need the iso to be installed as VM

lurch, 2023-03-05:
Did you mean `img` rather than `iso` ?
If you want to download one of the `.img` files "by hand", start at https://downloads.raspberrypi.com/os_list_imagingutility_v3.json to work out what the relevant URL is.

And see also #516

mirko8054, 2023-03-05:
No, i actually mean the ISO of the Ubuntu desktop Pi for installation on a VM, has to be a proper ISO with installation wizard, pretty much as amd64 version. Problem is Pi imager makes you directly write the SD card but i dont need it, i just need the iso

lurch, 2023-03-05:
Raspberry Pi Imager only uses IMG files, not ISO files. I wonder if you're looking for https://ubuntu.com/download/raspberry-pi ?

mirko8054, 2023-03-05:
YES how come i didnt find it in the website, could not find any related to raspberry at all, thanks

maxnet, 2023-03-05:
>Raspberry Pi Imager only uses IMG files, not ISO files. I wonder if you're looking for 
>https://ubuntu.com/download/raspberry-pi ?

If running under a HARDWARE VIRTUALIZED environments you'll probably want generic ARM ISOs instead of Pi specific stuff.
It is unlikely qemu/kvm/etc is emulating or passing thru Pi specific hardware...

So like:

https://cdimage.ubuntu.com/releases/22.10/release/ubuntu-22.10-live-server-arm64.iso
https://cdimage.debian.org/debian-cd/current/arm64/iso-cd/

If using CONTAINER BASED virtualization instead, you generally need to upload a tarball with the root file system to your virtualization software.
Like the RPI OS root.tar.xz that used to be there for NOOBS, but I don't think that is updated anymore...

Perhaps you have more luck with the competition, e.g.:

https://github.com/manjaro-arm/rootfs/releases

mirko8054, 2023-03-05:
Yeah Pimox is Hardware Virtualized, its nothing related to raspberry once i make a VM, this is why i need a regular desktop image and i cannot use preinstalled versions, seems like that every raspberry version on ubuntu.com is preinstalled, so it doesnt work. I assume an ubuntu desktop arm64 does not exists?

maxnet, 2023-03-05:
> I assume an ubuntu desktop arm64 does not exists?

It does:

https://cdimage.ubuntu.com/jammy/daily-live/current/jammy-desktop-arm64.iso

Not sure how useful it is to run desktop OSes under virtualization though.
They are quite resource intensive.

closed 2023-03-14 by ghollingworth