raspberrypi/rpi-imager #563: Unable to boot 3B+ from 64 bit raspios lite burned using rpi-imager

London-Lass, 2023-03-09:
Assets:-

- Pi : 3B+ (I have two)
- Desired Image : Raspberry PI OS  Lite (64-BIT)
- SD Card: Several different brands and sizes, 
- Card Reader :  uSD card adapter plugged into Mac M1 Pro / Lexar USB3 card reader plugged into either PC  Windows 11 or Mac M1 Pro

What works (boots, and usable) :- 

-  RPI -> Raspberry PI OS Lite (32-BIT) -> SD Card -> 3B+

What does not work (rainbow screen and 7 flashes on led)?

- RPI -> Raspberry PI OS Lite (64-BIT) -> SD Card -> 3B+

I've tried all the different permutations between, as listed, and consistently I cannot get a pi to boot when I burn the 64 bit version. Since the only thing i change in the process is the selection of which image to burn, I can't see where I am going wrong.

To eliminate a corrupt/cached build, I downloaded a couple of earlier 64 bit builds from the archive site, but same result.

London-Lass, 2023-03-09:
I've just realised that they are 2014 B+ not 2018 3B+. I presume this is the issue, in terms of compatibility?

maxnet, 2023-03-09:
Yes, older Pi does not understand 64-bit aarch64 instructions.

closed 2023-03-09 by maxnet

pelwell, 2023-03-09:
Indeed. 7 short flashes indicates "kernel not found". The firmware knows that kernel8.img, the only kernel binary in a 64-bit build, requires ARMv8 support and therefore won't run on a B+.