raspberrypi/rpi-imager #565: OS Request:  Add Rocky Linux Images

drushtx, 2023-03-13:
I posed this issue on the Rocky dev forum this week. No one has taken up the mantle to pursue requesting that Rocky images be included in the Raspberry Pi Imager. The Sr. Dev (Skip Grube) asked "anyone" to reach out so I took up the challenge.

I don't speak officially on behalf of Rocky/CIQ but I am a starting contact.

Rocky 8 and Rocky 9 images (downstream bug-for-bug clones to RHEL) are located at:  https://rockylinux.org/download

I'll be glad to answer any questions that I'm capable of. Thereafter, I will refer them to skip.

drushtx, 2023-03-14:
The image files are ready to go as .iso/img files.
Is there a dev that can respond to this request?  Thank you.

drushtx, 2023-03-14:
Thank you, @ghollingworth. I'll start researching the json file structure, contents. Appreciate any pointouts (I'm not a dev or coder - just volunteer who offered to get the process started).

tdewey-rpi, 2023-11-17:
Please fill out the form to request an addition to the OS list at https://forms.gle/cbsU27aLLVaxUXbw8.

closed 2023-11-17 by tdewey-rpi