raspberrypi/rpi-imager #568: Advanced options -> Set locale settings destroys /boot/config.txt and other /boot files

jalapenopuzzle, 2023-03-22:
Raspberry Pi Imager v1.7.4
Tested on Windows with image 2023-02-21-raspios-bullseye-arm64.img.xz
Tested with Raspberry Pi 4 Model B 4/8Gb

1. Write a MicroSD card using Raspberry Pi Imager. Select the following options:
    * Advanced Options -> Set locale settings
    * Time zone: Australia/Sydney
    * Keyboard layout: us
(Other time zones / keyboard layouts not tested - the critical setting is enabling Set locale settings)

2. Insert the MicroSD card into a Raspberry Pi
3. Wait for it to boot up and log in
4. Examine the file /boot/config.txt

With "Set locale settings" selected, the first 2048 bytes of the file /boot/config.txt are set to 00.
`hexdump /boot/config.txt`
```
0000000 0000 0000 0000 0000 0000 0000 0000 0000
*
0000800 230a 5220 6e75 6120 2073 6166 7473 6120
0000810 2073 6966 6d72 6177 6572 2f20 6220 616f
0000820 6472 6120 6c6c 776f 0a73 7261 5f6d 6f62
0000830 736f 3d74 0a31 5b0a 6c61 5d6c 000a     
000083d
```

Without "Set locale settings" selected, the file /boot/config.txt is 83 lines long and full of text.

The file size remains the same at 2109 bytes in both cases.

The corruption of the file /boot/config.txt has subtle and annoying effects that prevent all kinds of stuff from working.

I've been tearing my hair out all day trying to pin down this maddening issue. I'm looking forward to seeing a fix, but at least I know now NOT to use the "Set locale settings" setting.

maxnet, 2023-03-22:
Unable to reproduce.

>Insert the MicroSD card into a Raspberry Pi

You should also be able to access the FAT partition from Windows.
So if you do **not** put the card in the Pi, but reinsert it in your Windows computer, and you just open config.txt in notepad, is the file already corrupted at that stage?

jalapenopuzzle, 2023-03-22:
Yes I have observed the config.txt to be corrupt when mounted from Linux. 
Presumably it could be observed from Windows as well.

The complete list of settings I was using are as follows (without the 
specific values...)
1) Set host name
2) Enable SSH with password authentication
3) Set user name and password
4) Set locale settings

I'll do some more testing and see if I can isolate all the *necessary* 
contributing factors.

maxnet, 2023-03-22:
>Yes I have observed the config.txt to be corrupt when mounted from Linux.

If by "from Linux" you mean "on the Pi" instead of on a separate Linux desktop computer, there is the problem that the moment you boot up the Pi for the first time it will be doing a lot of writes to the SD card to resize partitions and file system to fill the capacity of your SD card.
It will also touch the FAT partition. The partition resize code will be removing something from cmdline.txt
And the customization script Imager installed will also be touching cmdline.txt, and remove firstrun.sh after consumed.

So then it is hard to tell if it was Imager that corrupted the file while doing the customizations.
Or the writes that are done on first boot by the Pi caused it. (If the latter is the case, you may also want to try a different brand of SD card if you have one).
That's why I am interested if you can already see the problem if you open the file in Windows, right after the write, **without the card having been in the Pi first**.

>The complete list of settings I was using are as follows (without the
>specific values...)
>1) Set host name
>2) Enable SSH with password authentication
>3) Set user name and password
>4) Set locale settings

But you only mention setting the timezone in your initial post. Why?
You didn't have a problem when setting 1+2+3, but only with 1+2+3+4?
Were you able to reproduce the problem more than once?

maxnet, 2023-03-22:
>I can mount the boot partition from an Ubuntu laptop (having just written the image) and I can observe the issue with 
>config.txt.

Also if you run Imager on that Ubuntu laptop to do the writing, instead of from Windows?

>I can reproduce this issue every time when setting all the options. This is not an intermittent fault. It depends 100% on 
>the settings used.

Cannot reproduce with all options, resulting in a 3025 bytes firstrun.sh
Is yours much bigger?

```
max@lynx:/media/max/bootfs$ cat issue.txt 
Raspberry Pi reference 2023-02-21
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 25e2319effa91eb95edd9d9209eb9f8a584d67be, stage2
max@lynx:/media/max/bootfs$ ls -l *.txt *.sh
-rw-r--r-- 1 max max  257 Mar 22  2023 cmdline.txt
-rw-r--r-- 1 max max 2109 Feb 21 04:53 config.txt
-rw-r--r-- 1 max max 3025 Mar 22  2023 firstrun.sh
-rw-r--r-- 1 max max  145 Feb 21 05:08 issue.txt
max@lynx:/media/max/bootfs$ hexdump config.txt |head
0000000 2023 6f46 2072 6f6d 6572 6f20 7470 6f69
0000010 736e 6120 646e 6920 666e 726f 616d 6974
0000020 6e6f 7320 6565 230a 6820 7474 3a70 2f2f
0000030 7072 2e66 6f69 632f 6e6f 6966 7467 7478
0000040 230a 5320 6d6f 2065 6573 7474 6e69 7367
0000050 6d20 7961 6920 706d 6361 2074 6564 6976
0000060 6563 6620 6e75 7463 6f69 616e 696c 7974
0000070 202e 6553 2065 696c 6b6e 6120 6f62 6576
0000080 6620 726f 6420 7465 6961 736c 0a0a 2023
0000090 6e75 6f63 6d6d 6e65 2074 6669 7920 756f
```

>I am currently testing whether I can reproduce the issue with any combination of fewer options, but each test takes time.

May want to test with RPI OS Lite 64-bit, instead of full 64-bit.
FAT partition of lite and full should be the same.

jalapenopuzzle, 2023-03-22:
I am writing the image and testing it from Ubuntu 22.04.
I have determined that all 4 settings are necessary to reproduce the problem, but also that some specific values are required to trigger the problem.
Now working on trying to isolate which values need to be specific.

maxnet, 2023-03-22:
>but also that some specific values are required to trigger the problem.

Specific values or specific length of those values?
Is the file size of firstrun.sh that gets created something special when the problem occurs?
E.g. exact multiple of 512 or 2048 bytes?

jalapenopuzzle, 2023-03-22:
I have a set of innocuous parameters which reproduces the problem.

Set Host Name: goode
Enable SSH: Ticked
SSH Password Authentication: Selected
Set User Name and Password: Ticked
User Name: branston
Password: password
Set locale settings: Ticked
Time Zone: Australia/Sydney
Keyboard Layout: us
Play sound when finished: no
Eject media when finished: yes
Enable telemetry: yes

Result: /boot/config.txt is corrupted with 2048 zeros at the front of the file.
This is not the only configuration which produces this result.
Other configurations have the same number of characters in the following fields:
Set Host Name
User Name
Password

I have not yet attempted to vary the number of characters in the fields to see if any particular field is important.

jalapenopuzzle, 2023-03-22:
In the corrupted state described above, /boot/firstrun.sh has size 2048.
Strangely the number of bytes that is corrupted in /boot/config.txt is also 2048 bytes.

jalapenopuzzle, 2023-03-22:
I have a second set of innocuous parameters which reproduces the problem.

Set Host Name: goode
Enable SSH: Ticked
SSH Password Authentication: Selected
Set User Name and Password: Ticked
User Name: branston
Password: passwor
Set locale settings: Ticked
Time Zone: Australia/Sydney
Keyboard Layout: us
Play sound when finished: no
Eject media when finished: yes
Enable telemetry: yes

Result: /boot/config.txt is corrupted with 2048 zeros at the front of the file.

In the corrupted state described above, /boot/firstrun.sh also has size 2048.

maxnet, 2023-03-22:
>Are you able to test with a different SD-card and/or a different card-reader?

Don't think that is the problem.
Think I made a very stupid bug in the FAT partition editing code in regards to rounding an integer division up. :-(

jalapenopuzzle, 2023-03-22:
> Are you able to test with a different SD-card and/or a different card-reader?

I did go down that path already and it is not the solution to this problem.

commit 65f6a07f (2023-03-22): FAT editing: fix creating files that are multiple of cluster size

Closes #568