raspberrypi/rpi-imager #569: Allow for extra partition

petermoo, 2023-03-22:
The Pi OS expands rootfs to use the rest of the microSD card. I sometimes add an "extras" partition at the end of the card. The Pi OS does not expand to use all the remaining space if there is any change to the card. I currently use Gparted to drag the end of rootfs up to the start of the extras partition. Perhaps Imager could perform the expand. Have an option to allocate x MB at the end for name abc then expand the rootfs partition to fill the reaming space.

maxnet, 2023-03-22:
Tools to resize ext4 partitions are not readily available under Windows and OS X.

If you have a lot of devices to install.
Do note that the software we provide to mass provision CM4 modules does offer an option for this: https://github.com/raspberrypi/cmprovision

petermoo, 2023-03-22:
Thank you for the feedback. I am happy to close this. Another reason why everyone should switch from closed proprietary OS. :-)

closed 2023-03-22 by petermoo