raspberrypi/rpi-imager #570: Error writing to storage on Rpi imager v1.7.4

Jasoniver, 2023-03-24:
Error writing to storage on Rpi imager v1.7.4, trying to install Mainsail from the drop downlist, it gets to 87% and then give the error writing to storage as seen below+

[image]
[image]

maxnet, 2023-03-24:
Tried a different SD card reader + card?

Jasoniver, 2023-03-24:
i have tried a different SD card but dont have another sd card reader
might go buy a reader tomorrow any suggestions?

i have read on some old posts about using a usb2.0 port works, any comments on that?

maxnet, 2023-03-24:
If the card reader has buggy firmware that does not get along with USB 3, a port that only support 2 may help.

If you get another reader, I do would get an USB 3 reader, though.
Not because the transfer speed makes much difference, but because the USB 3 readers tend to have newer chipsets inside.

Jasoniver, 2023-04-17:
Turns out it was my SD Card reader that didnt like my usb 3.1 port, bought a new reader and working fine now

closed 2023-04-17 by Jasoniver