raspberrypi/rpi-imager #571: When the wireless LAN is configured, the ethernet adapter is disabled.

sotnyk-lv, 2023-03-24:
Hi!
While configuring the Ubuntu server, I noticed that if the wireless LAN is configured in the Advanced options popup, the wired connection is disabled. 

It is unclear behavior.

The network configuration file is erased and replaced with something like this:
```
    wifis:
        renderer: networkd
        wlan0:
            access-points:
                SSID:
                    password: PASSKEY
            dhcp4: true
            optional: true
```
https://github.com/raspberrypi/rpi-imager/blob/9e882b31c48f92870750154b144dff17c5fbfe9b/src/OptionsPopup.qml#L738-L749

Shouldn't those settings be appended to the default file, or some other options be added to remain the eth0 connection active?

maxnet, 2023-03-25:
>Shouldn't those settings be appended to the default file

Not sure what behavior you are expecting.
Imager has no way of knowing what your plans and use-case are.
E.g. if Ethernet is going to be connected to a private network, while you use wlan for Internet connectivity.
And if so, if you want a dhcp client enabled for the private network, or if it is actually the Pi that will play DHCP server for your private network, in which case you certainly want a static network configuration.

Or if there is only one single network, and you expect Ethernet to be take precedence if a wired connection is plugged in, and wlan if it is not.
If the latter, be aware that if you just append wlan0 to the same network configuration that already has eth0, and you connect the Ethernet cable, while wlan for the same network is also reachable, it will result in undefined behavior which network interface will be used each time a TCP connection is made.
Which is probably not what you want. :-)

My personal preference is to create a bond0 active-backup bonding network interface that encapsulates both interface for such use-cases.
(I know there are others advocating different methods, such as appending both interfaces to the same configuration but giving them different routing metrics to indicate preference, but that may not always work. E.g. it may cause packets with the wlan IP as source address going over the Ethernet interface. Which may be accepted by your home network router, but is not going to fly in corporate networks that have DHCP snooping or similar security measures enabled on their Ethernet switches that drop packets that have a source address they are not expecting to come from that port).
But such network configurations are really beyond the scope of Imager.

sotnyk-lv, 2023-03-25:
Thank you for quick reply!

Yes, my case is that both interfaces lead to the same network, and eth0 is just preferred.
If I'm correct, if two networks are set in the same network configuration, they are given the default metrics. And as one is ethernet and the other is wireless, metrics are different, and eth0 is preferred. 

But I'm not confident in networking, and I'm not sure about it. 
Also, I can see your concerns about the different interface and IP.

Thank you for clearing that up!

closed 2023-03-25 by sotnyk-lv