raspberrypi/rpi-imager #572: COG icon does not appear after nominating OS (1.7.4)

SkipPack, 2023-03-26:
Specifically, I am writing an image of MoOde (2023-03-14-moode-r830-arm64-lite.img) to an SD card, but the COG does not appear so I can specify a login and password. While the image runs fine on my Pi, I need command line access for file system operations, not to mention security.

TIA,  Skip

SkipPack, 2023-03-26:
Should have mentioned -- I am using a 5 year  old I5 deskktop with up-to-date Win10

moodeaudio, 2023-03-26:
Hi Skip,

I'm able to repro this. Please reference this issue
https://github.com/raspberrypi/rpi-imager/issues/543

-Tim

maxnet, 2023-03-26:
>I'm able to repro this.

Imager is bad at guessing which type of configuration information we can feed your Linux distribution.
So will only offer customization if you advertise what your OS supports.
If your distribution is based on RPI OS try adding to your json:

```
"init_format": "systemd"
```

Or if your distribution supports cloud-init user-data files:

```
"init_format": "cloudinit"
```

moodeaudio, 2023-03-26:
Add to the item or subitems json?

moodeaudio, 2023-03-26:
Is there a similar json param to list in the "Media Player OS section"?

maxnet, 2023-03-26:
>nvm, its the subitems json and now the COG appears :-)

Good, then I'll close this.

closed 2023-03-26 by maxnet