raspberrypi/rpi-imager #573: Raspberry Pi Imager 1.7.4 generate a 64-bit image when a 32-bit image is selected

rkeithbeal, 2023-03-27:
In the 1.7.4 imager, I click "CHOOSE OS" then select "Raspberry Pi OS (other)" then choose Raspberry Pi OS Lite (32-bit)."
The image creat4ed is a 64-bit image.
pi@GW-12:~ $ uname -a
Linux GW-12 6.1.19-v8+ #1637 SMP PREEMPT Tue Mar 14 11:11:47 GMT 2023 aarch64 GNU/Linux

maxnet, 2023-03-27:
Kernel now defaults to 64-bit,
Try "file -L /lib/ld-linux.so.3" and you'll see you did get 32-bit userspace.

This is not an Imager issue, so I'll close this.
Forum is a better place to discuss, like here: https://forums.raspberrypi.com/viewtopic.php?p=2094017

rkeithbeal, 2023-03-27:
The 64-bit image generated is Kernel 6.1.19.
I was going to try to make a 32-bit Lite image for 6.1.19 manually but the download page 
https://www.raspberrypi.com/software/operating-systems/
shows 5.15 as the current version.

Raspberry Pi OS Lite

    Release date: February 21st 2023
    System: 32-bit
    Kernel version: 5.15
    Debian version: 11 (bullseye)
    Size: 362MB

rkeithbeal, 2023-03-30:
maxnet,

What do you mean by "But if you went through the first boot screen and gave it Internet access to let it install updates you do may end up with that, yes."

By the "first boot screen" do you mean just booting the SD?
What is the point of making a 32-bit image if it gets replaced o nthe first boot?

maxnet, 2023-03-30:
>By the "first boot screen" do you mean just booting the SD?

I mean that If you go through the first boot wizard thing that looks like the screenshots at: https://www.raspberrypi.com/news/raspberry-pi-bullseye-update-april-2022/
then you go through a screen in which it installs updates.

If you pre-configure username/password in Imager it may skip that though.
And I would expect updates to be skipped then as well, but that is a question that may be better to ask in the forum.
This repository is only about the Imager image writing program, and I am not involved with the OS images offered through it such as RPI OS, so not the right person to ask about specific details.

>What is the point of making a 32-bit image if it gets replaced o nthe first boot?

Userspace is still 32-bit, and is not replaced. Only kernel defaults to 64-bit after the update.
Whether it is right to call it "RPI OS 32-bit" is something you can debate over, but this is not the right place for that discussion.