raspberrypi/rpi-imager #574: reached end of fat32 root directory but no end-of-directory marker found - error on 1.7.4, works with 1.7.3

ciruz, 2023-03-28:
Hey,

I'm on Windows 11 and I tried to use Imager `v1.7.4` to install `Homebridge`, but I get following error:

`reached end of fat32 root directory but no end-of-directory marker found`

After downgrading to `v1.7.3`, it works without any problems.

I found the solution, with the downgrade tipp here: https://forums.raspberrypi.com/viewtopic.php?p=2094503#p2094503 , since somebody had the same problem there.

maxnet, 2023-03-28:
Already fixed in 43911005a21e57bfcaf92cbb4f547cd1ae3fc4ce

Misterbianchi, 2023-05-18:
I'm on version 1.7.4 and the issue persists, both on Win and mac PC.

I'm trying to install Homebridge.

Win 11: Version	10.0.22621 Build 22621
Mac: macOS Ventura 13.3.1 (a) (22772610a)

closed 2023-05-22 by maxnet